cherry-embedded/CherryUSB #12: ubuntu系统下，git clone后，直接git diff会提示CRLF 将被 LF 替换

jianjunjiang, 2022-02-08:
warning: class/audio/usbd_audio.c 中的 CRLF 将被 LF 替换。<
在工作区中该文件仍保持原有的换行符。
warning: class/cdc/usbd_cdc.c 中的 CRLF 将被 LF 替换。<
在工作区中该文件仍保持原有的换行符。
warning: class/cdc/usbh_cdc_acm.c 中的 CRLF 将被 LF 替换。<
在工作区中该文件仍保持原有的换行符。
warning: class/cdc/usbh_cdc_acm.h 中的 CRLF 将被 LF 替换。<
在工作区中该文件仍保持原有的换行符。
warning: class/hid/usbd_hid.c 中的 CRLF 将被 LF 替换。<
在工作区中该文件仍保持原有的换行符。
warning: class/hid/usbh_hid.c 中的 CRLF 将被 LF 替换。<
在工作区中该文件仍保持原有的换行符。
warning: class/hid/usbh_hid.h 中的 CRLF 将被 LF 替换。<
在工作区中该文件仍保持原有的换行符。
warning: class/hub/usbh_hub.c 中的 CRLF 将被 LF 替换。<
在工作区中该文件仍保持原有的换行符。
warning: class/hub/usbh_hub.h 中的 CRLF 将被 LF 替换。<
在工作区中该文件仍保持原有的换行符。
warning: class/msc/usbd_msc.c 中的 CRLF 将被 LF 替换。<
在工作区中该文件仍保持原有的换行符。
warning: class/msc/usbh_msc.c 中的 CRLF 将被 LF 替换。<
在工作区中该文件仍保持原有的换行符。
warning: class/msc/usbh_msc.h 中的 CRLF 将被 LF 替换。<
在工作区中该文件仍保持原有的换行符。
warning: class/video/usbd_video.c 中的 CRLF 将被 LF 替换。<
在工作区中该文件仍保持原有的换行符。
warning: common/usb_dc.h 中的 CRLF 将被 LF 替换。<
在工作区中该文件仍保持原有的换行符。
warning: common/usb_def.h 中的 CRLF 将被 LF 替换。<
在工作区中该文件仍保持原有的换行符。
warning: common/usb_util.h 中的 CRLF 将被 LF 替换。<
在工作区中该文件仍保持原有的换行符。
warning: core/usbd_core.c 中的 CRLF 将被 LF 替换。<
在工作区中该文件仍保持原有的换行符。
warning: core/usbd_core.h 中的 CRLF 将被 LF 替换。<
在工作区中该文件仍保持原有的换行符。
warning: core/usbh_core.c 中的 CRLF 将被 LF 替换。<
在工作区中该文件仍保持原有的换行符。
warning: core/usbh_core.h 中的 CRLF 将被 LF 替换。<
在工作区中该文件仍保持原有的换行符。
warning: usb_config.h 中的 CRLF 将被 LF 替换。<
在工作区中该文件仍保持原有的换行符。

sakumisu, 2022-02-08:
$ git config --global core.autocrlf false    # 关闭自动转换

jianjunjiang, 2022-02-08:
关闭后，强制复位，还剩 usb_config.h这个文件，提示文件内容变化。感觉还是应该配置到工程目录下，而不依赖全局配置环境。

sakumisu, 2022-02-08:
已修复

jianjunjiang, 2022-02-08:
主线已完美解决此问题，关闭此ISSUE

closed 2022-02-08 by jianjunjiang