cherry-embedded/CherryUSB #13: provide a way to override default SCSI_inquiry name

QIANWC, 2022-03-01:
here is the hardcoded define in usbd_msc.c:461

```
static bool SCSI_inquiry(uint8_t **data, uint32_t *len)
{
    //...some code

    uint8_t inquiry[SCSIRESP_INQUIRY_SIZEOF] = {
        /* 36 */

        /* LUN 0 */
        0x00,
        0x80,
        0x02,
        0x02,
        (SCSIRESP_INQUIRY_SIZEOF - 5),
        0x00,
        0x00,
        0x00,
        'B', 'o', 'u', 'f', 'f', 'a', 'l', 'o', /* Manufacturer : 8 bytes */
        'P', 'r', 'o', 'd', 'u', 'c', 't', ' ', /* Product      : 16 Bytes */
        ' ', ' ', ' ', ' ', ' ', ' ', ' ', ' ',
        '0', '.', '0', '1' /* Version      : 4 Bytes */
    };

```
Thanks.

sakumisu, 2022-03-03:
Yep，you are right，how about add macros in `usb_config.h` like `CONFIG_MSC_MANUFACTURER`?

QIANWC, 2022-03-05:
Any style you like is ok.
But I have to say that usb_config.h currently seems not to be user-editable but only overrideable.
Defining a simple value in project preprocessors is easy, but defining a long struct is not so compatible.
I suggest making a really customizable usb_config
eg:
#ifndef CHERRYUSB_USER_CONFIG_HEADER
#include "usb_config.h"
#else
#include CHERRYUSB_USER_CONFIG_HEADER
#end

sakumisu, 2022-03-05:
Hi, @QIANWC , usb_config.h in the root directory is just a templete, how this file built  is largely dependent on the user setting his header file path.,So if user wants to use their own usb_config.h,just includes their own header file path,and then usb_config.h in the root directory does not work.Thanks.

sakumisu, 2022-03-05:
For example is here : https://github.com/sakumisu/CherryUSB/tree/master/demo/stm32/usb_host/stm32f429igt6/Core/Inc

QIANWC, 2022-03-05:
Sorry, I misunderstood it. But actually I do use a custom usb_conf.h in my project.

sakumisu, 2022-03-05:
Thank you for your suggestion，any other question you can give me PR，i need it.

sakumisu, 2022-04-13:
Done

closed 2022-04-13 by sakumisu