cherry-embedded/CherryUSB #14: Can you please point me to a BL702 Host example [HID]?

rog77, 2022-03-11:
HI,

I'm interested in using this library in order to utilise a USB keyboard with a BL702 acting as the host, with a view to outputting keystrokes over uart as ascii - is there an example of such functionality that already exists that you might please point me to? Thanks!

sakumisu, 2022-03-12:
Hello，BL702 does not support host, the coming chips will support.Thank you!You can use other chips that i have supported.

rog77, 2022-03-12:
Thank you for your response, it looks like I must have misread/misinterpretated the data sheet for that chip; which is a shame as I needed to use that particular chip, as that is the one on the board I wish to adapt... Ah well.

closed 2022-03-18 by sakumisu