cherry-embedded/CherryUSB #18: 支持 UAC 2.0吗？

fyyxxm, 2022-03-30:


sakumisu, 2022-03-30:
支不支持是描述符说了算的，协议栈里面只管 class request

fyyxxm, 2022-03-30:
哦

sakumisu, 2022-04-08:
demo已经提上去了，你可以看下，6通道的mic

fyyxxm, 2022-04-08:
厉害！  谢谢！！

fyyxxm, 2022-04-08:
今天移植了下UAC1.0的，感觉是移植成功了：主要就是usb_dc_low_level_init的实现。
但是有些问题，电脑识别也不成功，用的V1的模板，什么都没动 OUT_EP也试了0x01。

AUDIO_OUT_EP 应该是0x01吧
[image]

感觉主要就是数据长度问题，
比如#define AUDIO_SIZEOF_AC_FEATURE_UNIT_DESC(ch, n) (7 + ch * n)  感觉应该是 (7 + （ch + 1）* n)
把这个改了 usb tree会多识别到Device Descriptor描述

USB Tree的信息：

    =========================== USB Port7 ===========================

Connection Status        : 0x02 (Device failed enumeration)
Port Chain               : 2-7
Properties               : 0x01
 IsUserConnectable       : yes
 PortIsDebugCapable      : no
 PortHasMultiCompanions  : no
 PortConnectorIsTypeC    : no
ConnectionIndex          : 0x07 (Port 7)
CompanionIndex           : 0
 CompanionHubSymLnk      : USB#ROOT_HUB30#5&1e232d6a&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
 CompanionPortNumber     : 0x03 (Port 3)
 -> CompanionPortChain   : 2-3

Device Manager Problem   : 43 (CM_PROB_FAILED_POST_START)
Used Endpoints           : 0

      ======================== USB Device ========================

        +++++++++++++++++ Device Information ++++++++++++++++++
Device Description       : 未知 USB 设备(设备描述符请求失败)
Device ID                : USB\VID_0000&PID_0002\6&35466DE&0&7
Hardware IDs             : USB\DEVICE_DESCRIPTOR_FAILURE
Driver KeyName           : {36fc9e60-c465-11cf-8056-444553540000}\0015 (GUID_DEVCLASS_USB)
Driver Inf               : C:\WINDOWS\inf\usb.inf
Legacy BusType           : PNPBus
Class                    : USB
Class GUID               : {36fc9e60-c465-11cf-8056-444553540000} (GUID_DEVCLASS_USB)
Enumerator               : USB
PDO                      : \Device\USBPDO-5
Location Info            : Port_#0007.Hub_#0001
Manufacturer Info        : (标准 USB 主控制器)
Capabilities             : 0x64 (Removable, SilentInstall, RawDeviceOK)
Status                   : 0x01806400 (DN_HAS_PROBLEM, DN_DISABLEABLE, DN_REMOVABLE, DN_NT_ENUMERATOR, DN_NT_DRIVER)
Problem Code             : 43 (CM_PROB_FAILED_POST_START)
Address                  : 7
HcDisableSelectiveSuspend: 0
EnableSelectiveSuspend   : 0
SelectiveSuspendEnabled  : 0
EnhancedPowerMgmtEnabled : 0
IdleInWorkingState       : 0
WakeFromSleepState       : 0
Power State              : D3 (supported: D0, D3, wake from D0)

        ---------------- Connection Information ---------------
Connection Index         : 0x07 (Port 7)
Connection Status        : 0x02 (DeviceFailedEnumeration)
Current Config Value     : 0x00 (Configuration 0)
Device Address           : 0x00 (0)
Is Hub                   : 0x00 (no)
Device Bus Speed         : 0x01 (Full-Speed)
Number Of Open Pipes     : 0x00 (0 pipes to data endpoints)
Data (HexDump)           : 07 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
                           00 00 00 00 00 00 00 01 00 00 00 00 00 00 00 02   ................
                           00 00 00                                          ...

        --------------- Connection Information V2 -------------
Connection Index         : 0x07 (7)
Length                   : 0x10 (16 bytes)
SupportedUsbProtocols    : 0x03
 Usb110                  : 1 (yes, port supports USB 1.1)
 Usb200                  : 1 (yes, port supports USB 2.0)
 Usb300                  : 0 (no, port not supports USB 3.0) -> but Companion Port 2-3 does
 ReservedMBZ             : 0x00
Flags                    : 0x00
 DevIsOpAtSsOrHigher     : 0 (Device is not operating at SuperSpeed or higher)
 DevIsSsCapOrHigher      : 0 (Device is not SuperSpeed capable or higher)
 DevIsOpAtSsPlusOrHigher : 0 (Device is not operating at SuperSpeedPlus or higher)
 DevIsSsPlusCapOrHigher  : 0 (Device is not SuperSpeedPlus capable or higher)
 ReservedMBZ             : 0x00
Data (HexDump)           : 07 00 00 00 10 00 00 00 03 00 00 00 00 00 00 00   ................

    ---------------------- Device Descriptor ----------------------
bLength                  : 0x00 (0 bytes)

      -------------------- String Descriptors -------------------
String descriptors are not available  (because the device has problem code CM_PROB_FAILED_POST_START)

sakumisu, 2022-04-09:
template里面无需任何改动，包括VID PID，你就算改成ST的，那也是走的标准的audio驱动。其次端点为什么要是0x01？这是谁说的，有很多芯片不支持双向啊。

sakumisu, 2022-04-09:
关于FEAUTURE UNIT中的ch，表示个数，不能填0，最小为1，你那个公式是可以填0的，但是误导性太强（因为audio手册是这么写的）

sakumisu, 2022-04-09:
可以加下qq群详细说吧：642693751

closed 2022-04-09 by fyyxxm