cherry-embedded/CherryUSB #20: Synopsys/EHCI/OHCI support

Devilliv, 2022-04-08:
@sakumisu 您好，有几个问题请教一下：
1）CherryUSB软件包是否是独立于RT-Thread原生自带的USB驱动框架，即单独打开软件包即可使用，不需要打开RT-Thread的components/drivers/usb？
2）CherryUSB中所支持的Synopsys IP，是DesignWare HS OTG Controller吗？若是，是否已支持高速host和高速device？
3）CherryUSB中已支持EHCI，但未支持OHCI，是否意味着类似全志的SoC中支持标准EHCI/OHCI的USB host暂时无法使用全速/低速设备（如USB鼠标/键盘等）？

希望能得到解答，谢谢！

sakumisu, 2022-04-09:
1、CherryUSB不依赖任何驱动框架 ，包括厂家驱动也不依赖
2、是 HS，但是没有支持 HS，需要外挂高速 PHY，所以只能工作在 FS下
3、个别芯片内部虽然是 EHCI,但是会加一些电路来支持 OHCI，但是表面上是 EHCI，所以，不清楚全志这个是什么情况，需要测试了才知道
4、OHCI测试过了会上传

Devilliv, 2022-04-09:
感谢解答！关于2）Synopsys IP 还想追问下：
我粗略浏览了一下代码，看到是有用CONFIG_USB_HS来区分支持HS的，是否可以理解为驱动是支持高速的，但需要硬件外挂ULPI HS PHY，而默认情况下芯片内置的UTMI FS PHY只能跑全速？

sakumisu, 2022-04-09:
是的，所以还有一个 CONFIG_USB_HS_IN_FULL用来在 hs下使用 full speed

Devilliv, 2022-04-09:
明白了，感谢！

Devilliv, 2022-04-26:
@sakumisu 您好，
我粗略查看了Synopsys Host的实现（usb_hc_synopsys.c），里面有用到stm32的HAL HCD driver的数据结构和API，请问是要依赖于stm32的底层实现吗？

sakumisu, 2022-04-26:
@Devilliv 没有精力更新该项，不好意思。有兴趣可以帮忙重构

sakumisu, 2022-04-26:
可以加群详细说，642693751

Devilliv, 2022-04-26:
已加，初学usb中，还要多多向作者学习

closed 2022-04-26 by Devilliv