cherry-embedded/CherryUSB #29: Later roadmap

sakumisu, 2022-05-14:
- **class driver**
- [ ] msc device with rt device interface，use thread process
- [ ] printer class support, device and host
- [ ] mtp class support, device and host
- [ ] cdc rndis additions 
- [ ] hub process optimise in host mode

- **port**
- [ ] synopsys dwc2 device port optimise
- [ ] synopsys dwc2 host port optimise
- [ ] renesas usb ip port
- [ ] rp2040 port ？
- [ ] ch32v307 host port
- [ ] ohci port
- [ ] ehci with iso transfer
- [ ] xhci port
- [ ] ftdi simulation，Copyright？

- **application**
- [ ] daplink template for all chips
- [ ] network
- [ ] uf2
- [ ] usb2 all bus：uart、i2c、spi、can
- [ ] dfu download with st tool

- **tool**
- [ ] ci test
- [ ] config tool update：winusb desc、hid report desc、audio desc、video desc
- [ ] class Stress tests scripts
- [ ] winusb windows and linux driver

aozima, 2022-05-14:
前排支持

geniusgogo, 2022-05-16:
666

HaiMianBBao, 2022-05-24:
期待更多应用的支持🉑

juiceRv, 2022-06-02:
完善了dwc2的支持，esp32 s2/s3的支持也可以做上了。。。

sakumisu, 2022-06-04:
> 完善了dwc2的支持，esp32 s2/s3的支持也可以做上了。。。还有GD32VF103........

是的，目前需要做的是对比寄存器的差距

Dandjinh, 2022-11-10:
application能否加入模拟ch340(usbfs)和ch348(usbhs)，可以参考wch开放的linux驱动，这样可以直接利用现用的已签名驱动。
目前我也正在研究ch340的linux驱动，可惜没这个基础，读起来比较费劲。

sakumisu, 2022-11-10:
> application能否加入模拟ch340(usbfs)和ch348(usbhs)，可以参考wch开放的linux驱动，这样可以直接利用现用的已签名驱动。 目前我也正在研究ch340的linux驱动，可惜没这个基础，读起来比较费劲。

指的是设备枚举成ch340这样？这个需要考虑版权问题

Dandjinh, 2022-11-10:
> > application能否加入模拟ch340(usbfs)和ch348(usbhs)，可以参考wch开放的linux驱动，这样可以直接利用现用的已签名驱动。 目前我也正在研究ch340的linux驱动，可惜没这个基础，读起来比较费劲。
> 
> 指的是设备枚举成ch340这样？这个需要考虑版权问题

驱动的license是GPL的，应该还好。
我看github上也有几个stm32模拟ch340的。

sakumisu, 2022-11-10:
> > > application能否加入模拟ch340(usbfs)和ch348(usbhs)，可以参考wch开放的linux驱动，这样可以直接利用现用的已签名驱动。 目前我也正在研究ch340的linux驱动，可惜没这个基础，读起来比较费劲。
> > 
> > 
> > 指的是设备枚举成ch340这样？这个需要考虑版权问题
> 
> 驱动的license是GPL的，应该还好。 我看github上也有几个stm32模拟ch340的。

这个我需要问下原厂，同意了才能做

AndyW999, 2023-04-01:
Looks like a very interesting project after all the problems I have had with the ST USB library.

My interest for RNDIS host to interface with Motorola and Hytera DMR radios.

I see RNDIS host is on your roadmap, please could you tell me if going be to be completed soon, as well as a way of integrating it with LWIP?

I could conduct some tests if that would help?

Thanks

Andy.

sakumisu, 2023-04-01:
> Looks like a very interesting project after all the problems I have had with the ST USB library.
>
> My interest for RNDIS host to interface with Motorola and Hytera DMR radios.
>
> I see RNDIS host is on [...]

It is completed, but using in rt-thread, and this is port: to use rndis api : https://github.com/sakumisu/CherryUSB/blob/master/third_party/rt-thread-4.1.1/rndis_host/rndis_host.c

AndyW999, 2023-04-06:
Is there going to be an RNDIS version that does not need an RTOS, or possibly using FreeRTOS as I do want to use it in the STM32 environment.

marchingband, 2023-05-18:
I would LOVE to see usb midi host support. I am working with BL616 for audio synthesis and midi host would be 💥

sakumisu, 2023-05-19:
> I would LOVE to see usb midi host support. I am working with BL616 for audio synthesis and midi host would be 💥

midi has no class request, when enumrates completely, go to user process.

marchingband, 2023-05-19:
@sakumisu I am not able to write this, because I am not experienced with USB protocol, is it possible to publish an example?

sakumisu, 2023-05-22:
> @sakumisu I am not able to write this, because I am not experienced with USB protocol, is it possible to publish an example?

I will give a simple demo for you this weekend.

marchingband, 2023-05-22:
@sakumisu  Thank you! If I am able to expand on it, I will offer to contribute back with a PR.

sakumisu, 2023-05-28:
> @sakumisu Thank you! If I am able to expand on it, I will offer to contribute back with a PR.

Use this driver to register midi class, and then use like cdc demo in usb_host.c to read or write midi device. I do not have midi device, but it is likely with cdc.

```
/*
 * Copyright (c) 2022, sakumisu
 *
 * SPDX-License-Identifier: Apache-2.0
 */
#include "usbh_core.h"
#include "usb_audio.h"

struct usbh_midi {
    struct usbh_hubport *hport;

    uint8_t intf;          /* interface number */
    usbh_pipe_t bulkin;  /* BULK IN endpoint */
    usbh_pipe_t bulkout; /* BULK OUT endpoint */
};

#define DEV_FORMAT "/dev/midi"

static int usbh_midi_connect(struct usbh_hubport *hport, uint8_t intf)
{
    struct usbh_endpoint_cfg ep_cfg = { 0 };
    struct usb_endpoint_descriptor *ep_desc;
    int ret;

    struct usbh_midi *midi_class = usb_malloc(sizeof(struct usbh_midi));
    if (midi_class == NULL) {
        USB_LOG_ERR("Fail to alloc midi_class\r\n");
        return -ENOMEM;
    }

    memset(midi_class, 0, sizeof(struct usbh_midi));
    midi_class->hport = hport;
    midi_class->intf = intf;

    hport->config.intf[intf].priv = midi_class;

    for (uint8_t i = 0; i < hport->config.intf[intf].altsetting[0].intf_desc.bNumEndpoints; i++) {
        ep_desc = &hport->config.intf[intf + 1].altsetting[0].ep[i].ep_desc;

        if (ep_desc->bEndpointAddress & 0x80) {
            usbh_hport_activate_epx(&midi_class->bulkin, hport, ep_desc);
        } else {
            usbh_hport_activate_epx(&midi_class->bulkout, hport, ep_desc);
        }
    }

    strncpy(hport->config.intf[intf].devname, DEV_FORMAT, CONFIG_USBHOST_DEV_NAMELEN);

    USB_LOG_INFO("Register MIDI Class:%s\r\n", hport->config.intf[intf].devname);

    return ret;
}

static int usbh_midi_disconnect(struct usbh_hubport *hport, uint8_t intf)
{
    int ret = 0;

    struct usbh_midi *midi_class = (struct usbh_midi *)hport->config.intf[intf].priv;

    if (midi_class) {
        if (midi_class->bulkin) {
            usbh_pipe_free(midi_class->bulkin);
        }

        if (midi_class->bulkout) {
            usbh_pipe_free(midi_class->bulkout);
        }

        if (hport->config.intf[intf].devname[0] != '\0') {
            USB_LOG_INFO("Unregister MIDI Class:%s\r\n", hport->config.intf[intf].devname);
        }

        memset(midi_class, 0, sizeof(struct usbh_midi));
        usb_free(midi_class);
    }

    return ret;
}

static const struct usbh_class_driver midi_streaming_class_driver = {
    .driver_name = "midi_streaming",
    .connect = usbh_midi_connect,
    .disconnect = usbh_midi_disconnect
};

CLASS_INFO_DEFINE const struct usbh_class_info midi_streaming_class_info = {
    .match_flags = USB_CLASS_MATCH_INTF_CLASS | USB_CLASS_MATCH_INTF_SUBCLASS,
    .class = USB_DEVICE_CLASS_AUDIO,
    .subclass = AUDIO_SUBCLASS_MIDISTREAMING,
    .protocol = 0x00,
    .vid = 0x00,
    .pid = 0x00,
    .class_driver = &midi_streaming_class_driver
};
```

marchingband, 2023-05-28:
@sakumisu thank you! Should `.protocol` `.vid` and `.pid` really be `0x00` or are those just placeholders? I really am new to this, sorry.

marchingband, 2023-05-28:
I have this file setting up for usb: https://github.com/marchingband/m0synth/blob/main/usb.c
I have this config: https://github.com/marchingband/m0synth/blob/main/usb_config.h
and in my main.c i call
```
void main(void){
    usbh_initialize();
    usb_midi_start();
    vTaskStartScheduler();
    while(1){}
}
```

and the log is simply `do not find /dev/midi`.

It's possible that there is a hardware issue (not enough capacitance to start the USB device), but I wan't to confirm I am using your library correctly. I followed the examples you suggested as close as I could.

Thanks again for your help!

sakumisu, 2023-05-29:
Please use usb_config.h in bouffalo_sdk, it is special.

marchingband, 2023-05-29:
that worked thank you 😎

marchingband, 2023-05-29:
@sakumisu I'm sorry I am stuck on a risc-v error that I am unfamiliar with, and have not been able to google much info about. Perhaps this is outside your driver, i don't know.
```
dynamic memory init success, ocram heap size = 181 Kbyte 
sig1:ffffffff
sig2:0000f32f
cgen1:9f7ffffd
Starting usb host task...
do not find /dev/midi
.[32m[I/USB] EHCI HCIVERSION:0100
.[0m.[32m[I/USB] EHCI HCSPARAMS:000001
.[0m.[32m[I/USB] EHCI HCCPARAMS:0006
.[0m.[32m[I/USB] New full-speed device on Hub 1, Port 1 connected
.[0m.[32m[I/USB] New device found,idVendor:09e8,idProduct:007c,bcdDevice:0100
.[0m.[32m[I/USB] The device has 2 interfaces
.[0m.[32m[I/USB] Enumeration success, start loading class driver
.[0m.[32m[I/USB] Loading audio_ctrl class driver
.[0m.[32m[I/USB] ============= Audio module information ===================
.[0m.[32m[I/USB] bcdADC :0100
.[0m.[32m[I/USB] Num of modules :0
.[0m.[32m[I/USB] Num of altsettings:1
.[0m.[32m[I/USB] ============= Audio module information ===================
.[0m.[32m[I/USB] Register Audio Class:/dev/audio0
.[0m.[32m[I/USB] Loading midi_streaming class driver
.[0mclass connectexception_entry
mcause=38000006
mepc:a000610c
mtval:0000400e
Store/AMO address misaligned
```
It seems to get to line 621 in usbh_core.c `ret = CLASS_CONNECT(hport, i);` and then the exception fires.
Please let me know if you have any ideas for me, and if not, thank you for your efforts here! I appreciate it!

sakumisu, 2023-05-29:
> @sakumisu I'm sorry I am stuck on a risc-v error that I am unfamiliar with, and have not been able to google much info about. Perhaps this is outside your driver, i don't know.
>
> ```
> dynamic memory [...]

Solve by yourself thanks, and do not reply here that nothing about roadmap

sakumisu, 2023-05-29:
> I am not able to debug the driver code, because I do not understand how it works 🙃. usb protocol is too complicated. I made the change you suggested but no change... tinyusb also does not have a working midi-host driver, so audio developers like me are searching for a way to accomplish it. If you have time later, and want to work on this demo, I am able to test on devices, write a midi parser, etc.. i am happy to contribute however I can. I know you are probably busy, and have many users asking for help. We appreciate you!

I have removed some comments,  if i have time to do this , i will tell you.

sakumisu, 2023-06-16:
> Is there going to be an RNDIS version that does not need an RTOS, or possibly using FreeRTOS as I do want to use it in the STM32 environment.

Support freertos, there is a demo in third-party

AndyW999, 2023-06-20:
Great!

How is the USB configuration set up as I could not see RNDIS as an option in your Configurator.

Thanks

Andy.

harbaum, 2023-08-04:
I see that the bl616 is supposed to be supported as a ehci  host. But the Buffalo port directory is empty.

Is there a cherryusb host example for the bl616?

sakumisu, 2023-08-05:
> I see that the bl616 is supposed to be supported as a ehci host. But the Buffalo port directory is empty.
> 
> Is there a cherryusb host example for the bl616?

https://github.com/bouffalolab/bouffalo_sdk/tree/master/examples/peripherals/usbhost

harbaum, 2023-08-07:
> https://github.com/bouffalolab/bouffalo_sdk/tree/master/examples/peripherals/usbhost

Thanks. I actually got the usbhost example to run on the m0s and it detects key presses from a USB keyboard. Excellent!

But the m0s really isn't comfortable to work with ...

alireza-tabatabaee, 2023-08-27:
Hi, I saw that the commits for the host video class (class/video/usbh_video.c and .h) are very recent and only 2 days ago. Does this mean the the host UVC class is still a work in progress? Or can it be used successfully in its current state?

sakumisu, 2023-08-27:
> Hi, I saw that the commits for the host video class (class/video/usbh_video.c and .h) are very recent and only 2 days ago. Does this mean the the host UVC class is still a work in progress? Or can it be used successfully in its current state?

It is done last year, but i do not provide for open source, only for business.

closed 2023-11-29 by sakumisu