cherry-embedded/CherryUSB #34: OTG over USB/IP

colemickens, 2022-06-02:
Hi!

I am _very excited_ to find this project.

I am trying to emulate USB MSD over USB/IP via ESP32. The end goal would be multiple Raspberry Pis, with ESP32's connected to their USB ports. The ESP32 would be available over USB/IP and then I would be able to use gadgetfs to configure the MSD to serve up over OTG.

Is this possible? I do not have experience with ESP-IDF, etc. I would be very thankful for any help, I might be able to donate some money if it would help too.

sakumisu, 2022-06-04:
Emulating esp32 with usb msd, then mount it over Raspberry Pis, right? Its ok. The thing to do now is to compare with esp32 usb ip with others, maybe use dwc2, i need sometime to review this port and optimise then.

sakumisu, 2022-09-23:
I review this issue, maybe we have some gaps, USB IP from mine means USB Intellectual Property, and from yours means Internet Protocol. And i guess you want to use tcp/ip with usb, like rndis class or cdc-ecm class. Right? Rndis supports.

closed 2022-10-07 by sakumisu