cherry-embedded/CherryUSB #35: Supported BSP for USBD_Video

wuhanstudio, 2022-06-04:
你好，

请问 USB Device 模拟摄像头现在除了 BL706_AVB 还支持其他开发板吗？似乎 BL706_AVB 只能在论坛申请。

https://dev.bouffalolab.com/media/doc/sdk/bl_mcu_sdk_zh/samples/advance%20samples/usb/usbd_video.html

sakumisu, 2022-06-04:
其他家的开发板没有引出特定的 dvp接口，所以没法做

sakumisu, 2022-06-04:
例程提供了简单的静态图发送的写法，不用 dvp 功能的

wuhanstudio, 2022-06-04:
多謝，這些支持列表裏我手邊只有一塊 F1C100S，不知道可不可以用 CherryUSB 静态图发送，

之前在 F1C100S 上試過 linux 的 configfs 驅動會報錯 invalid argument -22，大家說是 Endpoint 不夠：

https://bbs.aw-ol.com/topic/999/usb-gadget-%E6%A8%A1%E6%8B%9F%E6%91%84%E5%83%8F%E5%A4%B4-g_webcam-musb-hdrc-4-auto-failed-to-start-g_webcam-22/7

打算再買一塊開發板嘗試 USBD_Video，不知道下面這些開發板體哪一塊最方便呢？

[image]

sakumisu, 2022-06-04:
video的话，没有很大的 fifo或者dma搬运的 ip，基本没有用的意义，后续看下哪个合适再补充上来

sakumisu, 2022-06-04:
可以加入qq群一起交流：642693751

wuhanstudio, 2022-06-04:
> video的话，没有很大的 fifo或者dma搬运的 ip，基本没有用的意义，后续看下哪个合适再补充上来

多谢，我现在测试 i.mx6，树莓派 4，树莓派 zero 是支持的，全志系列的都不支持（H2 H3 H5 D1/D1s）

https://www.ridgerun.com/usb-videoclass-library-libguvc

sakumisu, 2022-06-05:
上面论坛说的端点问题，应该是不存在的，作为 device的话，没有端点不够这个问题的说法，作为host才有，其次可能 musb 他们家改的可能不支持吧，这个需要进行测试，audio测试是可以的

closed 2022-10-07 by sakumisu