cherry-embedded/CherryUSB #51: rt-thread环境中使用msh导出lsusb，使用lsusb查看枚举出的/dev/sda设备失败

mazy1991, 2022-09-24:
lsusb函数中有如下代码：
hport = &hub->child[port - 1];
若port = 0，则数组索引应该是溢出数值，修改为 hport = &hub->child[port]; 后lsusb可以正常查看到设备信息

sakumisu, 2022-09-24:
感谢，已修正

closed 2022-09-24 by sakumisu

mazy1991, 2022-11-11:
rt-thread环境下，usb_osal_sem_take接口内部实现有如下代码
int usb_osal_sem_take(usb_osal_sem_t sem, uint32_t timeout)
{
    int ret = 0;
    rt_err_t result = RT_EOK;

    result = rt_sem_take((rt_sem_t)sem, rt_tick_from_millisecond(timeout));
    if (result == RT_ETIMEOUT) {
        ret = -ETIMEDOUT;
    } else if (result == -T_ERROR) {
        ret = -EINVAL;
    } else {
        ret = 0;
    }

    return (int)ret;
}
此处条件判断语句有错，因为rt_sem_take返回结果小于等于零，所以代码应该修改如下
int usb_osal_sem_take(usb_osal_sem_t sem, uint32_t timeout)
{
    int ret = 0;
    rt_err_t result = RT_EOK;

    result = rt_sem_take((rt_sem_t)sem, rt_tick_from_millisecond(timeout));
    if (result == -RT_ETIMEOUT) {
        ret = -ETIMEDOUT;
    } else if (result == -RT_ERROR) {
        ret = -EINVAL;
    } else {
        ret = 0;
    }

    return (int)ret;
}

sakumisu, 2022-11-11:
谢谢，已修正