cherry-embedded/CherryUSB #63: ch32v307 cdc aсm

SeenTwo, 2022-11-09:
Tried your cdc acm example for my ch32v307 chip. On Linux, there are problems with registering a usb device:
```
[597460.594304] usb 1-6: new high-speed USB device number 68 using xhci_hcd
[597460.722349] usb 1-6: device descriptor read/64, error 18
[597460.958271] usb 1-6: device descriptor read/64, error 18
[597461.198248] usb 1-6: new high-speed USB device number 69 using xhci_hcd
[597461.326338] usb 1-6: device descriptor read/64, error 18
[597461.562305] usb 1-6: device descriptor read/64, error 18
[597461.670409] usb usb1-port6: attempt power cycle
[597462.086277] usb 1-6: new high-speed USB device number 70 using xhci_hcd
[597462.110555] usb 1-6: device descriptor read/8, error -61
[597462.242559] usb 1-6: device descriptor read/8, error -61
```

If you use CherryUSB from this [old repository](https://github.com/CherryUSB/cherryusb_ch32v307), then everything works:
```
[597604.323505] usb 1-6: new high-speed USB device number 72 using xhci_hcd
[597604.471899] usb 1-6: New USB device found, idVendor=ffff, idProduct=ffff, bcdDevice= 1.00
[597604.471905] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[597604.471909] usb 1-6: Product: CherryUSB CDC DEMO
[597604.471912] usb 1-6: Manufacturer: CherryUSB
[597604.471915] usb 1-6: SerialNumber: 2022123456
[597604.477337] cdc_acm 1-6:1.0: ttyACM1: USB ACM device
```

Why is this happening?

sakumisu, 2022-11-09:
please remove const in cdc acm descriptor,ch307  cannot visit flash with dma.

sakumisu, 2022-11-09:
I have tested, is ok

sakumisu, 2022-11-09:
And the old version ,all the data will copy to another ram then  use dma.

sakumisu, 2022-11-09:
I use this repo [cherryusb_ch32v307](https://github.com/CherryUSB/cherryusb_ch32v307) ,its ok

SeenTwo, 2022-11-09:
Thank you! From the old version, I still have this define `CONFIG_USBDEV_ALIGN_CHECK_DISABLE` in my configuration. Removed it and without `const` everything worked!

sakumisu, 2022-11-09:
Yes, `CONFIG_USBDEV_ALIGN_CHECK_DISABLE` is removed. All the port with dma must use align check.

closed 2022-11-09 by SeenTwo