cherry-embedded/CherryUSB #67: lsusb -v cdc device

SeenTwo, 2022-11-24:
Every time I do this on Linux:
`lsusb -v`

Then a log appears on my device:
```
[E/USB] descriptor <type:6,index:0> not found!
[E/USB] standard request error
[E/USB] standard request error
```

sakumisu, 2022-11-24:
It is impossible to print this log forever.

sakumisu, 2022-11-24:
If you use high speed, you must define `CONFIG_USB_HS`

sakumisu, 2022-11-24:
Please check with cdc_acm_template.c , it works, thank you! Try yourself again!

SeenTwo, 2022-11-28:
define CONFIG_USB_HS - yes. In IDE highlights what is defined. In build console shows: `-DCONFIG_USB_HS`.
Linux: kernel 5.4.0-132-generic, Kubuntu 20.04
Device: ch32v307.
string3 descriptor is uid device hex 24 symbols.

My descriptor:
```
#define VID	0x1A86
#define PID	0xFE0C

#define CDC_IN_EP  0x81
#define CDC_OUT_EP 0x02
#define CDC_INT_EP 0x83

#define USBD_MAX_POWER     100
#define USBD_LANGID_STRING 1033

/*!< config descriptor size */
#define USB_CONFIG_SIZE (9 + CDC_ACM_DESCRIPTOR_LEN)

/*!< global descriptor */
static uint8_t cdc_descriptor[] = {
    USB_DEVICE_DESCRIPTOR_INIT(USB_2_0, 0xEF, 0x02, 0x01, VID, PID, 0x0100, 0x01),
    USB_CONFIG_DESCRIPTOR_INIT(USB_CONFIG_SIZE, 0x02, 0x01, USB_CONFIG_BUS_POWERED, USBD_MAX_POWER),
    CDC_ACM_DESCRIPTOR_INIT(0x00, CDC_INT_EP, CDC_OUT_EP, CDC_IN_EP, 0x02),
    ///////////////////////////////////////
    /// string0 descriptor
    ///////////////////////////////////////
    USB_LANGID_INIT(USBD_LANGID_STRING),
    ///////////////////////////////////////
    /// string1 descriptor
    ///////////////////////////////////////
    0x08,                       /* bLength */
    USB_DESCRIPTOR_TYPE_STRING, /* bDescriptorType */
    'T', 0x00,                  /* wcChar0 */
    'T', 0x00,                  /* wcChar1 */
    'T', 0x00,                  /* wcChar2 */
    ///////////////////////////////////////
    /// string2 descriptor
    ///////////////////////////////////////
    0x14,                       /* bLength */
    USB_DESCRIPTOR_TYPE_STRING, /* bDescriptorType */
    'B', 0x00,                  /* wcChar0 */
    'B', 0x00,                  /* wcChar1 */
    'B', 0x00,                  /* wcChar2 */
    'B', 0x00,                  /* wcChar3 */
    'B', 0x00,                  /* wcChar4 */
    'B', 0x00,                  /* wcChar5 */
    'B', 0x00,                  /* wcChar6 */
    'B', 0x00,                  /* wcChar7 */
    'B', 0x00,                  /* wcChar8 */
    ///////////////////////////////////////
    /// string3 descriptor
    ///////////////////////////////////////
    0x32,                       /* bLength */
    USB_DESCRIPTOR_TYPE_STRING, /* bDescriptorType */
    0x00, 0x00,                 /* wcChar0 */
	0x00, 0x00,                 /* wcChar1 */
	0x00, 0x00,                 /* wcChar2 */
	0x00, 0x00,                 /* wcChar3 */
	0x00, 0x00,                 /* wcChar4 */
	0x00, 0x00,                 /* wcChar5 */
	0x00, 0x00,                 /* wcChar6 */
	0x00, 0x00,                 /* wcChar7 */
	0x00, 0x00,                 /* wcChar8 */
	0x00, 0x00,                 /* wcChar9 */
	0x00, 0x00,                 /* wcChar10 */
	0x00, 0x00,                 /* wcChar11 */
	0x00, 0x00,                 /* wcChar12 */
	0x00, 0x00,                 /* wcChar13 */
	0x00, 0x00,                 /* wcChar14 */
	0x00, 0x00,                 /* wcChar15 */
	0x00, 0x00,                 /* wcChar16 */
	0x00, 0x00,                 /* wcChar17 */
	0x00, 0x00,                 /* wcChar18 */
	0x00, 0x00,                 /* wcChar19 */
	0x00, 0x00,                 /* wcChar20 */
	0x00, 0x00,                 /* wcChar21 */
	0x00, 0x00,                 /* wcChar22 */
	0x00, 0x00,                 /* wcChar23 */
	0x00, 0x00,                 /* wcChar24 */
	0x00, 0x00,                 /* wcChar25 */
	0x00, 0x00,                 /* wcChar26 */
	0x00, 0x00,                 /* wcChar27 */
	0x00, 0x00,                 /* wcChar28 */
	0x00, 0x00,                 /* wcChar29 */
	0x00, 0x00,                 /* wcChar30 */
	0x00, 0x00,                 /* wcChar31 */
	0x00, 0x00,                 /* wcChar32 */
	0x00, 0x00,                 /* wcChar33 */
	0x00, 0x00,                 /* wcChar34 */
	0x00, 0x00,                 /* wcChar35 */
	0x00, 0x00,                 /* wcChar36 */
	0x00, 0x00,                 /* wcChar37 */
	0x00, 0x00,                 /* wcChar38 */
	0x00, 0x00,                 /* wcChar39 */
	0x00, 0x00,                 /* wcChar40 */
	0x00, 0x00,                 /* wcChar41 */
	0x00, 0x00,                 /* wcChar42 */
	0x00, 0x00,                 /* wcChar43 */
	0x00, 0x00,                 /* wcChar44 */
	0x00, 0x00,                 /* wcChar45 */
	0x00, 0x00,                 /* wcChar46 */
	0x00, 0x00,                 /* wcChar47 */
#ifdef CONFIG_USB_HS
    ///////////////////////////////////////
    /// device qualifier descriptor
    ///////////////////////////////////////
    0x0a,
    USB_DESCRIPTOR_TYPE_DEVICE_QUALIFIER,
    0x00,
    0x02,
    0x02,
    0x02,
    0x01,
    0x40,
    0x01,
    0x00,
#endif
    0x00
};
```

I also tried changing vid and pid. Problem still exists

SeenTwo, 2022-11-28:
I copied the cdc_acm_template.c example into my project, the problem is slightly different. Now when I call `lsusb -v`, I see one message in the debug messages of the device:
```[E/USB] standard request error```

On Linux, the output of lsusb has now changed:
```
Bus 001 Device 077: ID ffff:ffff CherryUSB CherryUSB CDC DEMO
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0xffff 
  idProduct          0xffff 
  bcdDevice            1.00
  iManufacturer           1 CherryUSB
  iProduct                2 CherryUSB CDC DEMO
  iSerial                 3 2022123456
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x004b
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass          2 Communications
      bFunctionSubClass       2 Abstract (modem)
      bFunctionProtocol       1 AT-commands (v.25ter)
      iFunction               0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      1 AT-commands (v.25ter)
      iInterface              2 CherryUSB CDC DEMO
      CDC Header:
        bcdCDC               1.10
      CDC Call Management:
        bmCapabilities       0x00
        bDataInterface          1
      CDC ACM:
        bmCapabilities       0x02
          line coding and serial state
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              16
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            2 Communications
  bDeviceSubClass         2 Abstract (modem)
  bDeviceProtocol         1 AT-commands (v.25ter)
  bMaxPacketSize0        64
  bNumConfigurations      1
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```

SeenTwo, 2022-11-28:
Is there something wrong with my descriptor?

SeenTwo, 2022-11-28:
I'm blind. The size of string3 was calculated incorrectly. Everything works now.
It remains to solve the problem: debug descriptor. But this topic is no longer relevant.

sakumisu, 2022-11-28:
Please disable other speed in device quality descriptor, `bNumConfigurations` should be 0.

SeenTwo, 2022-11-29:
Thank you, at least this problem is solved, I will think about another problem

closed 2022-11-29 by SeenTwo

sakumisu, 2022-11-29:
Thank you, if think this repo is good, devote me your star.