cherry-embedded/CherryUSB #80: cmsis-dap_v2.1_tempate.c

myshowtogo, 2023-01-02:
我正在stm32f103cb上制作一个daplink, 然后希望之后能平滑移植到其他mcu平台
我已经完成了cdc的功能, 已经可以连接msh
但是使用了cmsis-dap_v2.1_tempate.c之后出了一些问题
比如初始化的时候:
`[I/USB] Setup: bmRequestType 0x80, bRequest 0x06, wValue 0x0600, wIndex 0x0000, wLength 0x000a
[I/USB] Open endpoint:0x81 type:2 mps:64
[I/USB] Open endpoint:0x2 type:2 mps:64
[I/USB] Open endpoint:0x85 type:3 mps:8
[E/USB] Ep addr 133 overflow
[I/USB] Open endpoint:0x4 type:2 mps:64
[E/USB] Ep addr 4 overflow
[I/USB] Open endpoint:0x83 type:2 mps:64
[I/USB] get Compat ID
[I/USB] get Compat ID`
cdc我在使用cdc_acm_template.c时调试通过, 但是在使用daplink的cdc时,一旦pc连接串口,就会不停的上报错误:
`[E/USB] ep:85 clear halt
[E/USB] ep:85 clear halt
[E/USB] ep:85 clear halt
[E/USB] ep:85 clear halt
[E/USB] ep:85 clear halt
[E/USB] ep:85 clear halt
[E/USB] ep:85 clear halt
[E/USB] ep:85 clear halt`
断开串口的连接错误终止, 不知道是我哪里的使用有问题?
https://cherryusb.readthedocs.io/zh_CN/latest/index.html 在文档中也没有找到相关的介绍内容

sakumisu, 2023-01-03:
端点溢出了啊，你看报错的地方啊。

```
    if (ep_idx > (USB_NUM_BIDIR_ENDPOINTS - 1)) {
        USB_LOG_ERR("Ep addr %d overflow\r\n", ep_cfg->ep_addr);
        return -1;
    }
```

sakumisu, 2023-01-03:
肯定是你自己改了这个值的个数， ``USB_NUM_BIDIR_ENDPOINTS``

myshowtogo, 2023-01-03:
config里我修改了这个值, 我去除了这一行define, 库里面103默认是8, 现在可以正确使用dap的cdc了
但是依然有
[E/USB] standard request error
keil也无法识别到daplink设备, 还有哪里需要设置吗?

sakumisu, 2023-01-03:
这是硬件设计的，你为什么要改？

sakumisu, 2023-01-03:
``Setup: bmRequestType 0x80, bRequest 0x06, wValue 0x0600, wIndex 0x0000, wLength 0x000a`` 有这个错是正常的，这是高速命令，你一个全速当然不支持了，没有错才是不正常的

myshowtogo, 2023-01-03:
需要修改const uint8_t daplink_descriptor[] 吗?

sakumisu, 2023-01-03:
不需要，本来就是要用6个端点

myshowtogo, 2023-01-03:
> `Setup: bmRequestType 0x80, bRequest 0x06, wValue 0x0600, wIndex 0x0000, wLength 0x000a` 有这个错是正常的，这是高速命令，你一个全速当然不支持了，没有错才是不正常的

usb接触的时间比较短, 这个地方是要修改哪里?

sakumisu, 2023-01-03:
你是全速设备。。。你怎么改成高速？忽略就行啊，不是说了吗，没有错才是不正常，你现在报错了，不是很正常吗

sakumisu, 2023-01-03:
我只负责枚举成功，剩下的不归我这边管哈。

sakumisu, 2023-01-03:
https://github.com/HaiMianBBao/CherryUSB-Daplink

myshowtogo, 2023-01-03:
你这边的库我已经实现了msc读u盘, cdc, 确实性能比stm32自己做的库强很多, 公司这边的项目我现在也都用的你的库, 确实很厉害
daplink我这边已经用设备管理器识别到设备了, 但是keil识别不到

sakumisu, 2023-01-03:
你可以参考下我发的那个链接，跟现在的大差不差的，那个已经是一个模板了。

sakumisu, 2023-01-03:
> 你这边的库我已经实现了msc读u盘, cdc, 确实性能比stm32自己做的库强很多, 公司这边的项目我现在也都用的你的库, 确实很厉害 daplink我这边已经用设备管理器识别到设备了, 但是keil识别不到

好的感谢，更多问题可以加下readme底下qq群，有问题群里交流，会有更多人给你解答。

sakumisu, 2023-01-03:
觉得项目不错的话，可以贡献下你的 star 哈。

sakumisu, 2023-01-03:
简单测试方法，在 ``usb_dap_recv_callback`` 中加 log ，看看接收的数据和长度，以及发送是否ok。

myshowtogo, 2023-01-03:
star必须的, 之前没注意还有qq群, 这就进下

sakumisu, 2023-01-07:
默认 demo 没有问题 ，此 issue 就关了 。[image]

closed 2023-01-07 by sakumisu