cherry-embedded/CherryUSB #83: STM32f401. Failed to get device descriptor,errorcode:-116

ND-00, 2023-01-16:
I'm trying to make usb host on the stm32f401ccu6. This error occurs when connecting a usb device. I'm using CubeIDE. My code is copied from the stm32f429igt6 example. I followed the steps according to the guide.
That's all I did.

sakumisu, thanks for the great work! I hope the issue can be resolved

sakumisu, 2023-01-17:
You should check whether it supports dma mode.

ND-00, 2023-01-17:
MCU supports DMA mode. Or do you mean this?
[image]
My MCU does not have this option. Is it critical?
P.S. Sorry, maybe I'm asking very stupid questions, but I can't solve this problem for a long time. 
Error log
[image]

sakumisu, 2023-01-18:
Right, only support dma not fifo mode, and pin must be pb14,  pb15

ND-00, 2023-01-18:
The USB connector is configured and connected correctly. My pins are pb11, pb12. Do you have any ideas about this issue? I can't find the location of the problem

sakumisu, 2023-01-18:
sorry, must be another port with pin  PB14  PB15，PA11  do not support because it does not support dma

ND-00, 2023-01-18:
This MCU has no other outputs. I close the question. Thanks for the help!

closed 2023-01-18 by ND-00

sakumisu, 2023-01-18:
You can use stm32f429 and have a try.