cherry-embedded/CherryUSB #86: RT Thread + LwIP + RNDIS 例程

LeeChunHei, 2023-01-25:
请问作者有RT Thread + LwIP + RNDIS 的例程吗

我在RT Thread Studio建了CH32V307工程，使能了LWIP+CherryUSB CDC RNDIS
Ram, Flash 改成 128KB 192KB
然后用[cdc_rndis_template.c](https://github.com/sakumisu/CherryUSB/blob/master/demo/cdc_rndis_template.c)
可是运行后没有反应，按暂停时停在hardfault

sakumisu, 2023-01-25:
https://github.com/CherryUSB/CherryUSB_Device_RNDIS_Webserver

LeeChunHei, 2023-01-25:
有latest CherryUSB 的例子吗

sakumisu, 2023-01-25:
没有，自己debug吧，这应该是移植问题。

LeeChunHei, 2023-01-26:
如果CDC ACM 没有问题，bug 是不是在rndis 的代码里？

sakumisu, 2023-01-26:
有可能，这个挺久没更新的，用的人不是很多就没怎么管了，过完年我看下

sakumisu, 2023-01-26:
可以先看下枚举能不能过

sakumisu, 2023-01-28:
测试没有异常，stm32上

LeeChunHei, 2023-01-28:
用最新的CherryUSB？

sakumisu, 2023-01-28:
是的

LeeChunHei, 2023-01-28:
可以发工程文件给我看看吗
让我看看是不是我RT Thread 做错什么

sakumisu, 2023-01-28:
发不了

sakumisu, 2023-01-28:
反正告诉你不会异常就是了

sakumisu, 2023-01-28:
先能枚举再说吧，枚举都过不了，移植问题

LeeChunHei, 2023-01-28:
能枚举

sakumisu, 2023-01-28:
能枚举的话，剩下的就跟 usb 协议栈 没关系了，检查lwip跟rtt os了

LeeChunHei, 2023-01-28:
停在这里
[image]

LeeChunHei, 2023-01-28:
有点奇怪的是
ep 开了两次
[image]

sakumisu, 2023-01-28:
枚举过了，剩下的就不知道了

sakumisu, 2023-01-28:
只能自己调试了

sakumisu, 2023-01-28:
你可以把rtt相关的eth_device init全部屏蔽了，不让他对接到lwip

LeeChunHei, 2023-01-28:
如果eth_device_linkchange在
void usbd_configure_done_callback(void)
{
    rt_kprintf("config done\n");
    //eth_device_linkchange(&rndis_dev, RT_TRUE);
}
comment 掉
然后
void rt_usbd_rndis_init(void)
{
    rndis_dev.parent.control = rt_usbd_rndis_control;
    rndis_dev.eth_rx = rt_usbd_rndis_eth_rx;
    rndis_dev.eth_tx = rt_usbd_rndis_eth_tx;

    eth_device_init(&rndis_dev, "u0");

    eth_device_linkchange(&rndis_dev, RT_TRUE);
}
把  eth_device_linkchange(&rndis_dev, RT_FALSE); 改成  eth_device_linkchange(&rndis_dev, RT_TRUE);
不会hardfault
不过
rt_err_t rt_usbd_rndis_eth_tx(rt_device_t dev, struct pbuf *p)
{
    return usbd_rndis_eth_tx(p);
} 无论被叫多少次
void rndis_bulk_in(uint8_t ep, uint32_t nbytes) 也没有被调用过
然后USBHS_IRQHandler 也没有再中断

sakumisu, 2023-01-28:
我觉得跟usb 这边没什么关系，暂时就先这样吧

LeeChunHei, 2023-01-28:
我感觉有可能是ch32v307的移植问题
我有空再debug 看看
你用来测试的stm32 是哪款

sakumisu, 2023-01-28:
stm32f429，f407 都行

sakumisu, 2023-01-28:
你可以拉下最新的改动，估计 ch307 对全局变量初始化的内容没有做处理。

sakumisu, 2023-01-28:
当然了，这个不一定能解决。

sakumisu, 2023-01-28:
[image]

LeeChunHei, 2023-01-28:
还是不行，应该是移植问题了
我有空debug看看吧

sakumisu, 2023-01-28:
底下有qq群，可以加下。

sakumisu, 2023-01-28:
我这边反正啥也没改，用的rtt-master 下的bsp，勾两下就能用了

closed 2023-02-09 by sakumisu