cherry-embedded/CherryUSB #93: USB HID device stays in infinite loop when key pressed, when there is no USB support...

matsobdev, 2023-03-05:
...when plugged in to powered off computer (or restarting from working both computer and USB stack). After computer startup, stack initialises at BIOS USB and repetitive key pressing causes:
```
while (hid_state == HID_STATE_BUSY) {
}
in `hid_keyboard_template.c`
```
to loop to infinity and beyond.
Changing it to:
```
void send_report(uint8_t rep_id, uint8_t modifier, uint8_t key_code)
{
   uint8_t sendbuffer[9] = {rep_id, modifier, 0x00, key_code, 0x00, 0x00, 0x00, 0x00, 0x00};
   int ret = usbd_ep_start_write(HID_INT_EP, sendbuffer, 9);
   if (ret < 0) {
      return;
   }
   hid_state = HID_STATE_BUSY;
   // enough is enough - max time to proceed
   uint64_t start;
   uint64_t duration = 0;
   start = time_us_64();
   while (hid_state == HID_STATE_BUSY && duration < 100000) {
      duration = time_us_64() - start;
   }
}
```
solves the issue.

sakumisu, 2023-03-06:
It is user code, i just give a simple template.

matsobdev, 2023-03-06:
Yes, it's modified for that purpose of dealing with an issue. Everything's great, works nice but it can lock in described circumstances. Talking about unmodified `hid_keyboard_template.c`. It will be always `HID_STATE_BUSY ` then, so bypassing it to try again then it occurs too long.

sakumisu, 2023-03-07:
> Yes, it's modified for that purpose of dealing with an issue (only after comment). Everything's great, works nice but it locks in described circumstances. Talking about unmodified `hid_keyboard_template.c`. `hid_state` will be always `HID_STATE_BUSY` then, so bypassing it to try again when if occurs too long - I guess unnaturally long for HID keyboard.

This should be modified by users, i just tell that the api is ok, how to call the api is user work.

closed 2023-03-07 by matsobdev