cherry-embedded/CherryUSB #105: STM32H743 开启数据高速缓存后，USB DMA功能失效，会导致USB无法枚举

chengshuihang, 2023-05-08:
问题如题目所示。
原因如下
[image]
[image]
g_dwc2_udc未添加 USB_NOCACHE_RAM_SECTION宏，导致其被高速缓存命中，导致DMA枚举失败

sakumisu, 2023-05-08:
开了dma，这片内存不允许使用cache

chengshuihang, 2023-05-08:
是的，开了dma，这片内存不允许使用cache。但是当前usb_dc_dwc2.c文件中变量g_dwc2_udc没有使用宏USB_NOCACHE_RAM_SECTION修饰，需要手动在g_dwc2_udc变量上添加宏USB_NOCACHE_RAM_SECTION

sakumisu, 2023-05-08:
嗯嗯，可能之前写漏了，晚点我改下

chengshuihang, 2023-05-08:
将dwc2_reset函数   uint32_t count = 0U; 修改为 __IO uint32_t count = 0U;     修复在GCC环境下，因优化问题，未调用HAL库USB初始化，使用自定义的USB初始化函数，导致USB复位失败的bug。

sakumisu, 2023-05-08:
> 将dwc2_reset函数 uint32_t count = 0U; 修改为 __IO uint32_t count = 0U; 修复在GCC环境下，因优化问题，未调用HAL库USB初始化，使用自定义的USB初始化函数，导致USB复位失败的bug。

没看懂这个是什么意思。本来就不应该调用 hal库的usb代码，我都自己重写了，为什么还要调用他们的代码

chengshuihang, 2023-05-08:
就是不调用hal库的usb初始化代码，现在的dwc2_reset函数中uint32_t count = 0U，经过优化后，循环会立马跳出，导致USB复位失败，需要增加volatile关键字修饰count，保证不被优化，能够正确的完成USB复位动作

sakumisu, 2023-05-08:
不合理，这是局部变量，不会被优化

sakumisu, 2023-05-08:
建议使用armcc和clang

sakumisu, 2023-05-08:
好吧，会被优化成2000

closed 2023-05-09 by sakumisu