cherry-embedded/CherryUSB #111: usb_hc_dw2.c, dwc2_pipe_transfer, 第296到300行无意义

yangmafu, 2023-06-02:
第295行开始的代码如下：
```
    /* make sure to set the correct ep direction */
    if (ep_addr & 0x80) {
        tmpreg |= USB_OTG_HCCHAR_EPDIR;
    } else {
        tmpreg &= ~USB_OTG_HCCHAR_EPDIR;
    }

    /* Set host channel enable */
    tmpreg = USB_OTG_HC(ch_num)->HCCHAR;
    tmpreg &= ~USB_OTG_HCCHAR_CHDIS;
    tmpreg |= USB_OTG_HCCHAR_CHENA;
    USB_OTG_HC(ch_num)->HCCHAR = tmpreg;
```

此处需要配置HCCHAR寄存器中的EPDIR位。
代码逻辑中，先把EPDIR位的值写到变量tmpreg中，然后从HCCHAR寄存器中读取值到变量tmpreg。这导致前面写到变量tmpreg中的EPDIR配置被覆盖掉。也就是说296行到300行无意义。

改动如下：
```
    /* Set host channel enable */
    tmpreg = USB_OTG_HC(ch_num)->HCCHAR;
    tmpreg &= ~USB_OTG_HCCHAR_CHDIS;
    tmpreg |= USB_OTG_HCCHAR_CHENA;

    /* make sure to set the correct ep direction */
    if (ep_addr & 0x80) {
        tmpreg |= USB_OTG_HCCHAR_EPDIR;
    } else {
        tmpreg &= ~USB_OTG_HCCHAR_EPDIR;
    }

    USB_OTG_HC(ch_num)->HCCHAR = tmpreg;
```

sakumisu, 2023-06-02:
是的这边是有问题，应该是之前改错了

sakumisu, 2023-06-02:
已修正，感谢

closed 2023-06-02 by sakumisu