cherry-embedded/CherryUSB #118: STM32H743主机 USB枚举EC20有概率性会失败，控制器报告是DTERR，但是在USB总线上未观测到该错误

chengshuihang, 2023-07-14:
[image]
[image]
[image]
[image]

chengshuihang, 2023-07-14:
[image]
修改usb_hc_dwc2.c文件的两处位置，EC20可以稳定枚举。暂不清楚中断端点是否有影响

sakumisu, 2023-07-14:
不允许注释

sakumisu, 2023-07-14:
其次，你的分析仪太简陋了，自然分析不了这些错误，专业的分析仪是可以分析的

sakumisu, 2023-07-14:
对于STM32H7 还有其他需要注意的点，没有看到你的移植相关的内容，不好分析，可以加入readme下的群，进群说

closed 2023-07-15 by sakumisu