cherry-embedded/CherryUSB #119: USB端点传输异常后如何恢复

flzzaman, 2023-07-20:
应用场景：连接建立以后 USB设备会一直往host发送数据，发送的过程usb数据线有可能会被用户拔掉，再次插上uSB线后，需要枚举正常，所有端点工作正常，这种异常USB协议栈能自动处理恢复正常吗？目前提供的API 没有取消发送功能，如何USB断开以后 就不会调用发送结束回调。

sakumisu, 2023-07-20:
建议你先跑了代码再说吧。断开以后后续发送自动退出，不需要手动退出

sakumisu, 2023-07-20:
usbh_submit_urb usb断开以后自动退出，并且后续进入会一直退出的，不会进行传输

flzzaman, 2023-07-20:
usbh_submit_urb 是host协议栈中的发送函数吧？
我的应用是USB device，用的STM32H750芯片，采用的是ST自带的device库，使用中就发现了上述问题，如果host接收存在异常 device发数据后就会一直卡在那里，因此设计超时取消发送，释放互斥量；因为这个芯片测试发现不能多个端点同时发送，必须等一个端点发送完了，才能发送另一个端点的数据，发送上是做了互斥处理。如果一个端点异常了卡在那里，就会导致其他端点都没法发送；
目前是想用[CherryUSB]进行测试，发现接口很简单，所以存在疑问，是否会存在同样的问题，针对异常情况 应该如何处理？

sakumisu, 2023-07-20:
不存在不能同时发送，加超时是对的，但是取消是不太可能的，大部分ip没法手动取消，都是硬件自动取消

sakumisu, 2023-07-20:
如果host本身有问题，这也是需要解决的

flzzaman, 2023-07-20:
Host 是PC 按理不会有问题，但是长时间运行测试过程中，偶尔总会报usb io/pipe莫名奇妙的问题，设备不断电重启PC软件能恢复这是最基本要求了。如果不能手动不能取消，假如device向Host发送了一个50k长的数据包，发送到一半的时候这个管道出现异常，或者USB线被用户拔掉，这个时候device 这边会是什么行为呢

sakumisu, 2023-07-20:
自动停止

sakumisu, 2023-07-20:
你这是usb电路时序有问题，检查电路吧

flzzaman, 2023-07-20:
> 自动停止

大神能详细回答下吗， 会产生发送完成中断吗 还是只是暂停发送，再次建立连接以后会接着把剩余的发送完?

sakumisu, 2023-07-20:
如果是出错或者断开，则自动停止，不会继续发送，整个流程重来。就讲这么多，剩下的你自己研究，多谢

closed 2023-07-20 by sakumisu