cherry-embedded/CherryUSB #121: STM32F407 Problem

AndyW999, 2023-07-25:
I am using an STM32F407-Disco board using FS interface.
I have done all as described in https://cherryusb.readthedocs.io/zh_CN/latest/quick_start/stm32.html
I get the USB Serial Device enumerated in Win10 Device Manager but if I try to connect Putty to it I get 'Unable to open connection to COM3' 'Unable to configure serial port'

The STM32CubeIDE project is in Dropbox -

https://www.dropbox.com/scl/fo/ofwl3x4rvzbg2nj2xnqdy/h?rlkey=gdpsh40vuqq322hcixo89xakv&dl=0

Thanks

Andy.

sakumisu, 2023-07-25:
It is not usb issue i think, please refer to my stm32 demo, and confirm your board is ok.

AndyW999, 2023-07-25:
> Please check your printf, thanks, fputc is valid in keil not gcc.

I know ;)

I just copied it all.

AndyW999, 2023-07-25:
If I use the normal STM32CubeIDE ioc set up and use their middleware to set it to CDC it works and I can connect Putty.

So the board is OK.

I am using the cdc_acm_template.c file from your demo directory.

Viewing the USB status using USBDeview it shows ST Class 02, Protocol 01 with Cherry Class 00 Protocol 00. 

Please see attached files.

But as it is using your files I have no idea why.

[image]
[image]

sakumisu, 2023-07-26:
Please implment your printf, thanks, you do not implment this and COM3 is not cherryusb port.

sakumisu, 2023-07-26:
Others i cannot help you, try by yourself, or use my demo.

sakumisu, 2023-07-26:
https://github.com/CherryUSB/cherryusb_stm32

sakumisu, 2023-07-26:
If you do not implment printf, please do not use cherryusb

closed 2023-07-26 by sakumisu

AndyW999, 2023-07-26:
I have implemented printf and here are the results -

[I/USB] ========== dwc2 udc params ==========
[I/USB] CID:00001200
[I/USB] GSNPSID:4f54281a
[I/USB] GHWCFG1:00000000
[I/USB] GHWCFG2:229dcd20
[I/USB] GHWCFG3:020001e8
[I/USB] GHWCFG4:0ff08030
[I/USB] dwc2 fsphy type:1, hsphy type:0, dma support:0
[I/USB] dwc2 has 4 endpoints, default config: 4 endpoints
[I/USB] =================================
[W/USB] Ignore DQD in fs
[I/USB] Open ep:0x83 type:3 mps:8
[I/USB] Open ep:0x02 type:2 mps:64
[I/USB] Open ep:0x81 type:2 mps:64

sakumisu, 2023-07-26:
It works ok, but why serial port doesnot work, i dont know, sorry.

AndyW999, 2023-07-26:
If you have the time and patience to work with me I would like to solve this.

My goal is to eventually get RNDIS host working to interface with the DMR radios we use.

I still think the problem is related to the earlier views I sent from USBDeview  where the class and protocol are shown as 00

sakumisu, 2023-07-26:
If it is class and protocol issue, do you think this issue can  exist 2 years?

sakumisu, 2023-07-26:
Please use my demo, thanks ,others i cannot help you.

AndyW999, 2023-07-26:
I am using the code from your demo and following the instructions you give to do it.

I had to generate my own project as I do not have an STM32f469 board, only an STM32F407 board.

I have set up a Win7 32 bit VM and it will not even load a driver.

I tried a ‘clean’ Win10 VM and it took a while to get a Com port number but Putty kept timing out.

I have re-uploaded the STM32CubeIDE zipped archive project to the same Dropbox as before –

https://www.dropbox.com/scl/fo/ofwl3x4rvzbg2nj2xnqdy/h?rlkey=gdpsh40vuqq322hcixo89xakv&dl=0

sakumisu, 2023-07-26:
Solve it byyourself.

sakumisu, 2023-07-26:
Are you a chinese? You can join the qq group to talk with it.

AndyW999, 2023-07-26:
I have now tried NUCLEO -F429ZI board using FS and it is the same.

Please could you tell me which drivers are used in Windows 10?

Mine uses 
C:|WINDOWS\system32\DRIVERS\usbser.sys 
and I do not know if this is correct.

sakumisu, 2023-07-26:
And my class code is 02  02  01, not 00 00 00,thanks

sakumisu, 2023-07-27:
Nothing problem, sir, use USBDeview v3.06
[image]

sakumisu, 2023-07-27:
[image]

sakumisu, 2023-07-27:
[image]

sakumisu, 2023-07-27:
[image]

sakumisu, 2023-07-27:
Nothing problem, sir, use USBDeview v3.06

sakumisu, 2023-07-27:
And for stm32, they have their own windows driver, you can use other class like hid or msc. I doo not want to answer for this, it is an invalid time.

AndyW999, 2023-07-27:
Thanks for the screenshots!

This is really strange, I updated USBDview to 3.06, screenshot attached.

I did a driver update and put usbser.inf in it's own directory and told it to use it but it continues to use usb.inf.

Why do things like this always happen to me ;)

I have tried - 
STM32F407Disco with FS and HS
Nucleo STM32F429ZIT6 with FS

I do not know what to do as my other Windows PC's do the same.

Thanks for helping me, I do appreciate it ;)

[image]

AndyW999, 2023-07-27:
Is it possible for you to send me an STM32F407 working demo?

AndyW999, 2023-07-27:
And this is the output from UVCview.

[UVCView-CherryUSB_CDC.txt](https://github.com/cherry-embedded/CherryUSB/files/12180939/UVCView-CherryUSB_CDC.txt)

sakumisu, 2023-07-27:
And you do not use USBDview correctly, you should select COMX and device name is a serie number.

sakumisu, 2023-07-27:
You driver letter is null but mine is COM58.

sakumisu, 2023-07-27:
That's all, please check by yourself, if you think my descriptor is incorrect, you can copy st descriptor

sakumisu, 2023-07-27:
OK, i know, you can change your vid and pid to other, not ffff, you can modfiy with 0xabcd

sakumisu, 2023-07-27:
Last, you should solve by yourself, you can use other pc with windows or linux to have a try, or stop using cherryusb.

sakumisu, 2023-07-27:
It wastes me too much time.

AndyW999, 2023-07-27:
I know!

I have some STM32F103 boards soon, I will try it with them.

I know CherryUSB works but I wish it would for me.

Can you let me have a working set of source code that use have tried or even HEX code to program my STM32F407 board with?

I am sorry for wasting your time.

sakumisu, 2023-07-27:
how about testing other class like msc or hid？dwc driver is ok i think, maybe some others like windows drivers have confused

AndyW999, 2023-07-27:
Windows confused? However could that happen ;)

Yes, trying other classes was next for me to try but I will try not to bother you…

I will wait until I get the F103 boards so I can try the fsdev driver instead.