cherry-embedded/CherryUSB #122: 编译报错CherryUSB/class/wireless/usbd_rndis.c:17:40: error: 'memcpy' forming offset [129, 140] is out of the bounds [0, 128] of object 'rndis_encapsulated_resp_buffer' with type 'uint8_t[128]' {aka 'unsigned char[128]'} [-Werror=array-bounds]  #define RNDIS_INQUIRY_PUT(src, len)   (memcpy(infomation_buffer, src, len))

Kevincoooool, 2023-08-01:
CherryUSB/class/wireless/usbd_rndis.c:17:40: error: 'memcpy' forming offset [129, 140] is out of the bounds [0, 128] of object 'rndis_encapsulated_resp_buffer' with type 'uint8_t[128]' {aka 'unsigned char[128]'} [-Werror=array-bounds]
 #define RNDIS_INQUIRY_PUT(src, len)   (memcpy(infomation_buffer, src, len))

‘’‘switch (cmd->Oid) {
        case OID_GEN_SUPPORTED_LIST:
            RNDIS_INQUIRY_PUT(oid_supported_list, sizeof(oid_supported_list));
            infomation_len = sizeof(oid_supported_list);
            break;’‘’
这段代码是不是有问题

sakumisu, 2023-08-01:
没有问题多谢，自行关闭你的error

sakumisu, 2023-08-01:
或者自行修改 CONFIG_USBDEV_RNDIS_RESP_BUFFER_SIZE 大小

Kevincoooool, 2023-08-02:
> 或者自行修改 CONFIG_USBDEV_RNDIS_RESP_BUFFER_SIZE 大小

这个改成140可以了，还有请教一下RNDIS支持热插拔吗，我第一次上电时电脑可以获取到IP，但是重新插拔之后usbd_rndis_eth_rx就获取不到数据了

closed 2023-08-02 by Kevincoooool