cherry-embedded/CherryUSB #126: Using UVC example for ESP32-S3

marcpabst, 2023-08-06:
Hi, I'm trying to use the UVC exmaple in /demos on a ESP3-S3 board. WHile I can get some easier stuff ike HID input to work, I can't seem the camera class to function properly. When connecting to a Windows or Mac mashine, both detect a webcam with the correct name, but can't seem to receive any frames. Moreover, on Windows, the ESP32 is constantly resetting (resulting in the well-known Windows USB-connected sound). 

Maybe there is something I'm missing, but I can't seem to get it working. Any hints/tips?

Here is my `main.c`:
```c
#include <stdio.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "spi_flash_mmap.h"
#include "hal/usb_hal.h"
#include "soc/usb_periph.h"
#include "rom/gpio.h"
#include "driver/gpio.h"
#include "driver/periph_ctrl.h"
#include <stdio.h>

//#include "hid_mouse_template.c"
#include "video_static_mjpeg_template.c"

static void ConfigureUsbPins(usb_hal_context_t *usb)
{
    /* usb_periph_iopins currently configures USB_OTG as USB Device.
     * Introduce additional parameters in usb_hal_context_t when adding support
     * for USB Host.
     */
    for (const usb_iopin_dsc_t *iopin = usb_periph_iopins; iopin->pin != -1; ++iopin)
    {
        if ((usb->use_external_phy) || (iopin->ext_phy_only == 0))
        {
            gpio_pad_select_gpio(iopin->pin);
            if (iopin->is_output)
            {
                gpio_matrix_out(iopin->pin, iopin->func, false, false);
            }
            else
            {
                gpio_matrix_in(iopin->pin, iopin->func, false);
                gpio_pad_input_enable(iopin->pin);
            }
            gpio_pad_unhold(iopin->pin);
        }
    }
    if (!usb->use_external_phy)
    {
        gpio_set_drive_capability(USBPHY_DM_NUM, GPIO_DRIVE_CAP_3);
        gpio_set_drive_capability(USBPHY_DP_NUM, GPIO_DRIVE_CAP_3);
    }
}

void app_main()
{

    // TURNO LED ON GPIO2 ON
    gpio_pad_select_gpio(2);
    gpio_set_direction(2, GPIO_MODE_OUTPUT);
    gpio_set_level(2, 1);

    periph_module_reset(PERIPH_USB_MODULE);
    periph_module_enable(PERIPH_USB_MODULE);

    usb_hal_context_t hal = {
        .use_external_phy = false};
    usb_hal_init(&hal);
    printf("Configuring USB pins\n");
    ConfigureUsbPins(&hal);

    video_init();
    video_test();
}
```

marcpabst, 2023-08-06:
I managed to get this backtrace which should be helpful:

    Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 
    
    Core  0 register dump:
    PC      : 0x4200427f  PS      : 0x00060034  A0      : 0x82005345  A1      : 0x3fc92180  
    A2      : 0x00000001  A3      : 0x3fc96070  A4      : 0x00000010  A5      : 0x000003fc  
    A6      : 0x00000000  A7      : 0x3fc9a620  A8      : 0x3fc9a544  A9      : 0x00000010  
    A10     : 0x00000001  A11     : 0xff000000  A12     : 0x803770b1  A13     : 0x3fc9a5d0  
    A14     : 0x3fc9a5f0  A15     : 0x00000000  SAR     : 0x00000001  EXCCAUSE: 0x00000005  
    EXCVADDR: 0x00000000  LBEG    : 0x40056fc5  LEND    : 0x40056fe7  LCOUNT  : 0x00000000  
    Core  0 was running in ISR context:
    EPC1    : 0x4201db63  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4200427f
    
    
    Backtrace: 0x4200427c:0x3fc92180 0x42005342:0x3fc921a0 0x4200169b:0x3fc921d0 0x403766d1:0x3fc921f0 0x4200161c:0x3fc9a670 0x42001691:0x3fc9a6a0 0x4201eacf:0x3fc9a6d0 0x4037c735:0x3fc9a700
    
      #0  0x4200427c:0x3fc92180 in dwc2_tx_fifo_empty_procecss at main/usb_cherry/CherryUSB/port/dwc2/usb_dc_dwc2.c:423
      #1  0x42005342:0x3fc921a0 in OTG_FS_IRQHandler at main/usb_cherry/CherryUSB/port/dwc2/usb_dc_dwc2.c:1112
      #2  0x4200169b:0x3fc921d0 in usb_interrupt_cb at main/usb_cherry/addition/usb_cherry_platform.c:31
      #3  0x403766d1:0x3fc921f0 in _xt_lowint1 at /Users/user/.platformio/packages/framework-espidf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1118
      #4  0x4200161c:0x3fc9a670 in one_frame at main/video_static_mjpeg_template.c:250
      #5  0x42001691:0x3fc9a6a0 in app_main at main/main.c:78 (discriminator 1)
      #6  0x4201eacf:0x3fc9a6d0 in main_task at /Users/user/.platformio/packages/framework-espidf/components/freertos/FreeRTOS-Kernel/portable/port_common.c:131 (discriminator 2)
      #7  0x4037c735:0x3fc9a700 in vPortTaskWrapper at /Users/user/.platformio/packages/framework-espidf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:154
    
    
    Core  1 register dump:
    PC      : 0x403789fa  PS      : 0x00060334  A0      : 0x82008579  A1      : 0x3fc9b560  
    A2      : 0x00000000  A3      : 0x0000abab  A4      : 0x8037ca32  A5      : 0x3fc98330  
    A6      : 0x00060023  A7      : 0x00000000  A8      : 0x8201d10a  A9      : 0x3fc9b520  
    A10     : 0x00000000  A11     : 0x00000003  A12     : 0x00000008  A13     : 0x00001001  
    A14     : 0x0000000c  A15     : 0xfffffff6  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
    EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
    
    
    Backtrace: 0x403789f7:0x3fc9b560 0x42008576:0x3fc9b580 0x4037b195:0x3fc9b5a0 0x4037c735:0x3fc9b5c0
    
      #0  0x403789f7:0x3fc9b560 in xt_utils_wait_for_intr at /Users/user/.platformio/packages/framework-espidf/components/xtensa/include/xt_utils.h:81
          (inlined by) esp_cpu_wait_for_intr at /Users/user/.platformio/packages/framework-espidf/components/esp_hw_support/cpu.c:101
      #1  0x42008576:0x3fc9b580 in esp_vApplicationIdleHook at /Users/user/.platformio/packages/framework-espidf/components/esp_system/freertos_hooks.c:59
      #2  0x4037b195:0x3fc9b5a0 in prvIdleTask at /Users/user/.platformio/packages/framework-espidf/components/freertos/FreeRTOS-Kernel/tasks.c:4157 (discriminator 1)
      #3  0x4037c735:0x3fc9b5c0 in vPortTaskWrapper at /Users/user/.platformio/packages/framework-espidf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:154
    
    
    
    
    ELF file SHA256: 5bb9ed6838cc7952

sakumisu, 2023-08-07:
I do not use esp32s3, but video demo does not support for dwc2 fs port, bacase its fifo is too less, please use other chips. thanks.

marcpabst, 2023-08-07:
I see. I actually have a BL808 board in the mail. I'm just a bit concerned about the asymmetric architecture--do you know if it's possible to run USB stuff on the D0 core? I'm looking to receive data via the DVP interface and send it off via USB and I'm slightly worried it'll be tough without two cores!

sakumisu, 2023-08-07:
Because of fifo limitation, dwc2 fs usb ip has 1280 byte, but uvc demo need 1023 byte, of course you can modify 1023 to other size,(and you should modify CONFIG_USB_DWC2_TX1_FIFO_SIZE, default is 128) but it will be just a unuseful toy.

sakumisu, 2023-08-07:
DVP+UVC on BL808 is ok, you can refer to sipeed m1s dock sdk: https://github.com/sipeed/M1s_BL808_example/tree/main/c906_app/uvc_demo

sakumisu, 2023-08-10:
> Thanks! Just to understand, why would you suggest changing CONFIG_USB_DWC2_TX1_FIFO_SIZE?

I have said, uvc demo uses 1023 bytes, but dwc2 tx1 only support 128 bytes, it is overflow.

closed 2023-08-13 by sakumisu