cherry-embedded/CherryUSB #128: EHCI timeouts on BL616 as host

harbaum, 2023-08-12:
I do see many timeouts when using the USB host on a BL616. This can be "fixed" by adding a delay to EHCI usbh_submit_urb() like so:

```
    if (urb->timeout > 0) {
        usb_osal_msleep(1);
        /* wait until timeout or sem give */
        ret = usb_osal_sem_take(pipe->waitsem, urb->timeout);
        if (ret < 0) {
            goto errout_timeout;
        }
        ret = urb->errorcode;
    }
```

This makes the setup stable and e.g. running FATFS on top of MSC just works fine. But I am sure this delay is not a good solution and this will sure have a massive negative impact on performance.

I am seeing this on the current master, integrated into the latest bouffalo SDK.

Example usbhost with fatfs enabled. Without this delay:

```
  ____               __  __      _       _       _     
 |  _ \             / _|/ _|    | |     | |     | |    
 | |_) | ___  _   _| |_| |_ __ _| | ___ | | __ _| |__  
 |  _ < / _ \| | | |  _|  _/ _` | |/ _ \| |/ _` | '_ \ 
 | |_) | (_) | |_| | | | || (_| | | (_) | | (_| | |_) |
 |____/ \___/ \__,_|_| |_| \__,_|_|\___/|_|\__,_|_.__/ 

Build:19:15:52,Aug 12 2023
Copyright (c) 2022 Bouffalolab team
======== flash cfg ========
flash size 0x00400000
jedec id     0xC86016
mid              0xC8
iomode           0x04
clk delay        0x01
clk invert       0x01
read reg cmd0    0x05
read reg cmd1    0x35
write reg cmd0   0x01
write reg cmd1   0x01
qe write len     0x02
cread support    0x01
cread code       0x20
burst wrap cmd   0x77
===========================
dynamic memory init success, ocram heap size = 173 Kbyte 
sig1:ffffffff
sig2:0000f32f
cgen1:9f7ffffd
Starting usb host task...
starting scheduler
do not find /dev/ttyACM0
do not fi[I/USB] EHCI HCIVERSION:0100
[I/USB] EHCI HCSPARAMS:000001
[I/USB] EHCI HCCPARAMS:0006
t0 not finnd /dev/inpud /dev/sda

[I/USB] New high-speed device on Hub 1, Port 1 connected
[I/USB] New device found,idVendor:054c,idProduct:0243,bcdDevice:0100
[I/USB] The device has 1 interfaces
[I/USB] Enumeration success, start loading class driver
[I/USB] Loading msc class driver
[I/USB] Get max LUN:1
[I/USB] Ep=81 Attr=02 Mps=512 Interval=255 Mult=00
[I/USB] Ep=02 Attr=02 Mps=512 Interval=255 Mult=00
[I/USB] Capacity info:
[I/USB] Block num:7831552,block size:512
[I/USB] Register MSC Class:/dev/sda
do not find /dev/ttyACM0
do not find /dev/input0
do not find /dev/ttyACM0
data len:25
[I/USB] MSC_disk_read(0x22fc93c0,0,1)
[I/USB] MSC_disk_read(0x22fc93c0,62,1)
[I/USB] MSC_disk_read(0x22fc93c0,63,1)
test fatfs write
[I/USB] MSC_disk_read(0x22fc93c0,16446,1)
[I/USB] MSC_disk_read(0x22fc93c0,16447,1)
[I/USB] MSC_disk_read(0x22fc93c0,16448,1)
[I/USB] MSC_disk_write(0x22fc93c0,16448,1)
[I/USB] MSC_disk_read(0x22fc93c0,5496,1)
[I/USB] MSC_disk_write(0x22fc9a00,4423918,4)
write success, write len：2500
[I/USB] MSC_disk_write(0x22fc9600,4423922,1)
[I/USB] MSC_disk_write(0x22fc93c0,5496,1)
[I/USB] MSC_disk_write(0x22fc93c0,13123,1)
[I/USB] MSC_disk_read(0x22fc93c0,16448,1)
[I/USB] MSC_disk_write(0x22fc93c0,16448,1)
[I/USB] MSC_disk_write(0x22fc93c0,63,1)
[I/USB] MSC_disk_ioctl(0,(nil))
test fatfs read
[I/USB] MSC_disk_read(0x22fc93c0,16446,1)
[I/USB] MSC_disk_read(0x22fc93c0,16447,1)
[I/USB] MSC_disk_read(0x22fc93c0,16448,1)
[I/USB] MSC_disk_read(0x22fc9a00,4423918,4)
[I/USB] MSC_disk_read(0x22fc9600,4423922,1)
do not find /dev/ttyACM0
do not find /dev/input0
do not find /dev/ttyACM0
do not find /dev/input0
do not find /dev/ttyACM0
do not find /dev/ttyACM0
do not find /dev/input0
do not find /dev/ttyACM0
[E/USB] usbh_msc_bulk_in_transfer()=-116
[E/USB] msc data transfer error
read fail
```

With the delay:

```
  ____               __  __      _       _       _     
 |  _ \             / _|/ _|    | |     | |     | |    
 | |_) | ___  _   _| |_| |_ __ _| | ___ | | __ _| |__  
 |  _ < / _ \| | | |  _|  _/ _` | |/ _ \| |/ _` | '_ \ 
 | |_) | (_) | |_| | | | || (_| | | (_) | | (_| | |_) |
 |____/ \___/ \__,_|_| |_| \__,_|_|\___/|_|\__,_|_.__/ 

Build:19:15:52,Aug 12 2023
Copyright (c) 2022 Bouffalolab team
======== flash cfg ========
flash size 0x00400000
jedec id     0xC86016
mid              0xC8
iomode           0x04
clk delay        0x01
clk invert       0x01
read reg cmd0    0x05
read reg cmd1    0x35
write reg cmd0   0x01
write reg cmd1   0x01
qe write len     0x02
cread support    0x01
cread code       0x20
burst wrap cmd   0x77
===========================
dynamic memory init success, ocram heap size = 173 Kbyte 
sig1:ffffffff
sig2:0000f32f
cgen1:9f7ffffd
Starting usb host task...
starting scheduler
do not find /dev/ttyACM0
do not fi[I/USB] EHCI HCIVERSION:0100
[I/USB] EHCI HCSPARAMS:000001
[I/USB] EHCI HCCPARAMS:0006
t0 not finnd /dev/inpud /dev/sda

[I/USB] New high-speed device on Hub 1, Port 1 connected
[I/USB] New device found,idVendor:054c,idProduct:0243,bcdDevice:0100
[I/USB] The device has 1 interfaces
[I/USB] Enumeration success, start loading class driver
[I/USB] Loading msc class driver
[I/USB] Get max LUN:1
[I/USB] Ep=81 Attr=02 Mps=512 Interval=255 Mult=00
[I/USB] Ep=02 Attr=02 Mps=512 Interval=255 Mult=00
[I/USB] Capacity info:
[I/USB] Block num:7831552,block size:512
[I/USB] Register MSC Class:/dev/sda
do not find /dev/ttyACM0
do not find /dev/input0
do not find /dev/ttyACM0
data len:25
[I/USB] MSC_disk_read(0x22fc93c0,0,1)
[I/USB] MSC_disk_read(0x22fc93c0,62,1)
[I/USB] MSC_disk_read(0x22fc93c0,63,1)
test fatfs write
[I/USB] MSC_disk_read(0x22fc93c0,16446,1)
[I/USB] MSC_disk_read(0x22fc93c0,16447,1)
[I/USB] MSC_disk_read(0x22fc93c0,16448,1)
[I/USB] MSC_disk_write(0x22fc93c0,16448,1)
[I/USB] MSC_disk_read(0x22fc93c0,5496,1)
[I/USB] MSC_disk_write(0x22fc93c0,5496,1)
[I/USB] MSC_disk_write(0x22fc93c0,13123,1)
[I/USB] MSC_disk_read(0x22fc93c0,16448,1)
[I/USB] MSC_disk_read(0x22fc93c0,5496,1)
[I/USB] MSC_disk_write(0x22fc9a00,4423918,4)
write success, write len：2500
[I/USB] MSC_disk_write(0x22fc9600,4423922,1)
[I/USB] MSC_disk_write(0x22fc93c0,5496,1)
[I/USB] MSC_disk_write(0x22fc93c0,13123,1)
[I/USB] MSC_disk_read(0x22fc93c0,16448,1)
[I/USB] MSC_disk_write(0x22fc93c0,16448,1)
[I/USB] MSC_disk_write(0x22fc93c0,63,1)
[I/USB] MSC_disk_ioctl(0,(nil))
test fatfs read
[I/USB] MSC_disk_read(0x22fc93c0,16446,1)
[I/USB] MSC_disk_read(0x22fc93c0,16447,1)
[I/USB] MSC_disk_read(0x22fc93c0,16448,1)
[I/USB] MSC_disk_read(0x22fc9a00,4423918,4)
[I/USB] MSC_disk_read(0x22fc9600,4423922,1)
read success, read len：2500
```

sakumisu, 2023-08-13:
No need ,thanks,adding delay is a joke.

sakumisu, 2023-08-13:
Many echi cpu works fine, check it by yourself

closed 2023-08-13 by sakumisu

harbaum, 2023-08-13:
I am not sure I understand you correctly.

So the BL616 EHCI host is not supported and problems like this will not be addressed? And you suggest to use a different MCU or to fix any issues myself, correct?

Which MCU would you suggest to use with cherryUSB in host mode?

sakumisu, 2023-08-13:
No problem in BL616, and others like d1s/hpm6750 and so on with ehci.Just check by yourself.

harbaum, 2023-08-13:
I am checking this myself. This report is a result of tests with real hardware. I use a BL616 in host mode with various USB devices as I am going to decide if we build a device based on that chip 

Most USB host tests are unstable. Some very basic tests work. That's how I did the joystick demo:
 https://m.youtube.com/shorts/Z1DRsI0xd8o

But hubs, mass storage and even some joysticks run into timeouts. 

I am checking real hardware and would like to understand if the BL616 is supported in a way that we can base a project on it.

So what do you mean by "check it yourself" other than using the BL616 to do tests with real USB devices?

sakumisu, 2023-08-13:
I do not want to waste my time on these unofficial  board, no one knows if the board usb is ok, so solve by yourself.

harbaum, 2023-08-13:
> I do not want to waste my time on these unofficial board, no one knows if the board usb is ok, so solve by yourself.

Ah ... now we are getting somewhere. This is some internal conflict between Bouffalo and SiPeed. Interesting. So only Bouffalo's own dev kit is supported? This would mean that we cannot use the BL616 for our own designs, since there won't be any support then?

BTW: The delay actually doesn't hurt as much as expected. I copied your SD card test over to the USB demo and get these results with my delay in place:

```
[I/USB] EHCI HCIVERSION:0100
[I/USB] EHCI HCSPARAMS:000001
[I/USB] EHCI HCCPARAMS:0006
[I/USB] New high-speed device on Hub 1, Port 1 connected
[I/USB] New device found,idVendor:054c,idProduct:0243,bcdDevice:0100
[I/USB] The device has 1 interfaces
[I/USB] Enumeration success, start loading class driver
[I/USB] Loading msc class driver
[I/USB] Get max LUN:1
[I/USB] Ep=81 Attr=02 Mps=512 Interval=255 Mult=00
[I/USB] Ep=02 Attr=02 Mps=512 Interval=255 Mult=00
[I/USB] Capacity info:
[I/USB] Block num:7831552,block size:512
[I/USB] Register MSC Class:/dev/sda
[I/USB] ******************** be about to write test... **********************
[I/USB] Write Test Succeed! 
[I/USB] Single data size:32768 Byte, Write the number:1024, Total size:32768 KB
[I/USB] Time:9846ms, Write Speed:3328 KB/s 
[I/USB] ******************** be about to read test... **********************
[I/USB] Read Test Succeed! 
[I/USB] Single data size:32768Byte, Read the number:1024, Total size:32768 KB
[I/USB] Time:4289ms, Read Speed:7640 KB/s 
[I/USB] ******************** be about to check test... **********************
[I/USB] Check Test Succeed! 
[I/USB] All Data Is Good! 
```

3.3MBytes/s writing and 7.6 MBytes/s reading doesn't sound too bad for some cheap little microcontroller.

harbaum, 2023-08-13:
BTW: What _is_ the official board for the BL616? I see someone testing a BL616DK at https://codingfield.com/blog/2023-01/getting-started-with-bl618dk/, but that seems to be in fact a BL618 and I don't see where this can be purchased nor are there any schematics, so I could compare the USB implementation of the official board with the unofficial one from SiPeed.

sakumisu, 2023-08-13:
You can design sch and pcb by yourself, m0s dock is designed only for debugger.

harbaum, 2023-08-13:
I don't understand that. What kind of debugger?

sakumisu, 2023-08-13:
cklink, daplink, usb2uart

harbaum, 2023-08-13:
You can flash the M0S dock without any debugger and just flash it via USB as I explain e.g. at https://github.com/harbaum/Pacman-TangNano9k/tree/main/m0sdock_usb_joystick

Would you consider the [Ai-M62-12F](https://m.de.aliexpress.com/item/1005004486335583.html) a valid device and accept feedback for it?

I really like to test the stuff before designing a new device.

sakumisu, 2023-08-14:
I mean that M0s is only a usb device, its function is a debugger or a usb2uart, thanks, please stop for this issue.

harbaum, 2023-08-14:
I'll stop now. Just to make this clear: There is no BL616 based board that I could use and get cherryusb support for, right? If there is one, then please tell me. The BL616 seems to have much potential and it seems to be one of the few cheap chips that can do high speed USB. But without support this chip is of course way less useful.

sakumisu, 2023-08-26:
Hi, i have updated ehci driver just now, you can have a tryand copy them to sdk to see if it works or not. PS: if you want to use usb host, recommand you to use ai-thinker board.

harbaum, 2023-08-26:
Thanks a lot. I recently received the ai thinker boards and will give the updated version a try asap ...

harbaum, 2023-08-26:
I just tested on the M0S and now USB host isn't working at all, anymore. No events are reported. And adding my old delay also doesn't help.

I tried a) using the entire latest cherryusb from github master and b) using only the latest ehci from github master. This makes no difference. Neither works.

I'll try again on Monday using the AI Thinker board instead of the M0S.

Thanks for your support!

sakumisu, 2023-08-26:
I think it is your incorrect copying....

sakumisu, 2023-08-26:
please do not modify usb_config in bouffalo sdk, it is not a standard ehci.

sakumisu, 2023-08-26:
Ok, i know what is wrong, please modify CONFIG_USB_EHCI_HCOR_RESERVED_DISABLE in usb_config.h, in bouffalo_sdk it is CONFIG_USB_ECHI_HCOR_RESERVED_DISABLE

sakumisu, 2023-08-26:
If you have other problem, you can contact my twitter.

harbaum, 2023-08-26:
> Ok, i know what is wrong, please modify `CONFIG_USB_EHCI_HCOR_RESERVED_DISABLE` in usb_config.h, in bouffalo_sdk it is `CONFIG_USB_ECHI_HCOR_RESERVED_DISABLE`, it is a spelling error.

Ah! Yes, now it works without my delay patch. Great. The simple demos from the SDK now work.

I have ported the more complex fatfs test from the SD card demo to the USB host demo. I do see sporadic write errors in the big file there. The SD card has a very similar problem but when reading (writing seems to work fine with SD card). See https://github.com/bouffalolab/bouffalo_sdk/issues/170

Since the errors are very similar (chunks 32 bytes get lost during reading from SD card and during writing to USB stick) I think this is not an issue with USB, but with the BL616 in general. So I won't bother you with that.

>  If you have other problems, you can contact my twitter, maybe solve problems more quickly
I don't have a twitter account. Do you use discord?

sakumisu, 2023-08-26:
Sorry, i do not have and join for some reason. Maybe only in github, could you give my repo a star and push some issues here, only this way, feel sorry.

harbaum, 2023-08-26:
No problem. I gave this repo a star.

This is the code for the updated usb_host.c demo from the bouffalo SDK which will do extensive USB file testing and finally fail. I can open an issue for this (is this what you mean by "push some issues"?). 

But as I said, I see the same problem with the SD card, so I don't think it's a USB specific issue. It is either a problem with the SDK in general or with the BL616 itself.

```
#include "usbh_core.h"
#include "usbh_msc.h"

#include "ff.h"
#include "diskio.h"

USB_NOCACHE_RAM_SECTION FATFS fs;
USB_NOCACHE_RAM_SECTION FIL fnew;
UINT fnum;
FRESULT res_msc = 0;

int MSC_disk_status() { return 0; }
int MSC_disk_initialize() { return 0; }
int MSC_disk_read(BYTE *buff, LBA_t sector, UINT count) {
  struct usbh_msc *msc_class = (struct usbh_msc *)usbh_find_class_instance("/dev/sda");
  // USB_LOG_INFO("MSC_disk_read(%p,%d,%d)\r\n", buff, sector, count);  
  return usbh_msc_scsi_read10(msc_class, sector, buff, count);
}

int MSC_disk_write(const BYTE *buff, LBA_t sector, UINT count) {
  struct usbh_msc *msc_class = (struct usbh_msc *)usbh_find_class_instance("/dev/sda");
  // USB_LOG_INFO("MSC_disk_write(%p,%d,%d)\r\n", buff, sector, count);  
  return usbh_msc_scsi_write10(msc_class, sector, buff, count);
}

int MSC_disk_ioctl(BYTE cmd, void *buff) {
  USB_LOG_INFO("MSC_disk_ioctl(%d,%p)\r\n", cmd, buff);
  
    switch (cmd) {
        // Get R/W sector size (WORD)
        case GET_SECTOR_SIZE:
            break;

        // Get erase block size in unit of sector (DWORD)
        case GET_BLOCK_SIZE:
            break;

        case GET_SECTOR_COUNT:
            break;

        case CTRL_SYNC:
            break;

        default:
            break;
    }

    return 0;
}

DSTATUS Translate_Result_Code(int result)
{
    return result;
}

#define SDU_DATA_CHECK 1

char test_data[512] =
    "I've been reading books of old \r\n\
    The legends and the myths \r\n\
    Achilles and his gold \r\n\
    Hercules and his gifts \r\n\
    Spiderman's control \r\n\
[... 201 lines cut ...]
            usb_osal_msleep(2000);
            continue;
        }

        usb_msc_fatfs_test();

        while (1) {
            msc_class = (struct usbh_msc *)usbh_find_class_instance("/dev/sda");
            if (msc_class == NULL) {
                goto find_class;
            }
            usb_osal_msleep(2000);
        }
    }
}

void usbh_msc_run(struct usbh_msc *msc_class) { }

void usbh_msc_stop(struct usbh_msc *msc_class) { }

void usbh_class_test(void) {
  usb_osal_thread_create("usbh_msc", 2048, CONFIG_USBHOST_PSC_PRIO + 1, usbh_msc_thread, NULL);
}

```

sakumisu, 2023-08-27:
I think it is cache problem, and you should set USB_NOCACHE_RAM_SECTION USB_MEM_ALIGNX before read_buffer, at least usb host need.

sakumisu, 2023-08-27:
And i must say, all of buffer for usb, must be in nocache ram section, thanks.

sakumisu, 2023-08-27:
And MSC_disk_ioctl must be implemented.

harbaum, 2023-08-29:
Using a small no-cache transfer buffer makes the MSC demo works with full filesystem support and success in the 32MB read/write test.

Thanks a lot!

harbaum, 2023-08-29:
Using 

`bflb_l1c_dcache_clean_range((void *)(RW_Buffer), sizeof(RW_Buffer));
`
and

`bflb_l1c_dcache_invalidate_range((void *)(RW_Buffer), sizeof(RW_Buffer));`

works even better and has much less impact on the performance.

sakumisu, 2023-08-29:
Right, with cache, you must do these. But many users do not know cache, this is why exists nocache ram.

harbaum, 2023-08-29:
How long is a cache line on the BL616? Sometimes you align to 32 bytes (e.g. in the USB) and sometimes to 64 bytes (e.g. in SDH).

sakumisu, 2023-08-30:
> How long is a cache line on the BL616? Sometimes you align to 32 bytes (e.g. in the USB) and sometimes to 64 bytes (e.g. in SDH).

Cache line 32, so set 32 or 64 is ok, just you like.