cherry-embedded/CherryUSB #131: [dwc2 dc] EP0 is bi-direction?

wosayttn, 2023-08-18:
Hi,

These if conditional sentences are unnecessary in these function, right?
Or, we should ignore the EP0 checking?

```c
int usbd_ep_start_write(const uint8_t ep, const uint8_t *data, uint32_t data_len)
{
...
    if (USB_OTG_INEP(ep_idx)->DIEPCTL & USB_OTG_DIEPCTL_EPENA)
    {
        return -2;
    }
...
}

int usbd_ep_start_read(const uint8_t ep, uint8_t *data, uint32_t data_len)
{
...
    if (USB_OTG_OUTEP(ep_idx)->DOEPCTL & USB_OTG_DOEPCTL_EPENA)
   {
        return -2;
    }
...
}

```

sakumisu, 2023-08-18:
This checks if ep transfers done, it is necessary.

wosayttn, 2023-08-18:
> This checks if ep transfers done, it is necessary.
After appending "ignore ep_idx=0", the CONFIG_USB_DWC2_DMA_ENABLE mode is workable.

```c
int usbd_ep_start_write(const uint8_t ep, const uint8_t *data, uint32_t data_len)
{
    ....
    if (ep_idx && (USB_OTG_INEP(ep_idx)->DIEPCTL & USB_OTG_DIEPCTL_EPENA))
    {
        return -2;
    }
    ....
}
int usbd_ep_start_read(const uint8_t ep, uint8_t *data, uint32_t data_len)
{
    ...
    if (ep_idx && (USB_OTG_OUTEP(ep_idx)->DOEPCTL & USB_OTG_DOEPCTL_EPENA)) //Endpoint enable?
    {
        return -2;
    }
    ...
}
```

sakumisu, 2023-08-18:
Not adding works in mine.

sakumisu, 2023-08-18:
You can see my stm32 demo, using stm32f4 and stm32h7 with dma mode. It works.

sakumisu, 2023-08-18:
USB_OTG_DIEPCTL_EPENA will be set when start reading or writing and be cleared when transfers done.

sakumisu, 2023-08-18:
Certainly, this check is not necessary.

wosayttn, 2023-08-18:
Thanks comment, I will check other in our porting. CherryUSB is good implementation. Good job.

sakumisu, 2023-08-18:
Thanks, maybe this check can disable?

closed 2023-08-18 by wosayttn