cherry-embedded/CherryUSB #132: Static MJPEG example on M1S Dock

marcpabst, 2023-08-18:
Hi, I"m trying to run the UVC static MJPEG exmple on my. It compiles and flashes fine, and both Windows and MacOS recognize the camera seemingly correctty. However, I never get an image transferred. Here it the log from the serial monitor:

      ____               __  __      _       _       _     
     |  _ \             / _|/ _|    | |     | |     | |    
     | |_) | ___  _   _| |_| |_ __ _| | ___ | | __ _| |__  
     |  _ < / _ \| | | |  _|  _/ _` | |/ _ \| |/ _` | '_ \ 
     | |_) | (_) | |_| | | | || (_| | | (_) | | (_| | |_) |
     |____/ \___/ \__,_|_| |_| \__,_|_|\___/|_|\__,_|_.__/ 
    
    Build:23:50:55,Aug 18 2023
    Copyright (c) 2022 Bouffalolab team
    ======== flash cfg ========
    flash size 0x01000000
    jedec id     0xEF4018
    mid              0xEF
    iomode           0x04
    clk delay        0x01
    clk invert       0x01
    read reg cmd0    0x05
    read reg cmd1    0x35
    write reg cmd0   0x01
    write reg cmd1   0x31
    qe write len     0x01
    cread support    0x00
    cread code       0xFF
    burst wrap cmd   0x77
    ===========================
    dynamic memory init success,heap size = 21 Kbyte 
    sig1:ffff32ff
    sig2:0000ffff
    video test
    [I/USB] Close ep:0x81 type:1
    CLOSE
    CLOSE
    [I/USB] Close ep:0x81 type:1
    CLOSE
    CLOSE
    [E/USB] descriptor <type:3,index:4> not found!
    [E/USB] standard request error
    [I/USB] Setup: bmRequestType 0x80, bRequest 0x06, wValue 0x0304, wIndex 0x0409, wLength 0x00ff

sakumisu, 2023-08-21:
I do not know. Sorry, It works in mine.

sakumisu, 2023-08-21:
[image]

marcpabst, 2023-08-21:
You"re using this one https://github.com/CherryUSB/cherryusb_bouffalolab/tree/master/examples/peripherals/usbdev/usbd_static_video?

sakumisu, 2023-08-22:
Right

sakumisu, 2023-08-22:
> You"re using this one https://github.com/CherryUSB/cherryusb_bouffalolab/tree/master/examples/peripherals/usbdev/usbd_static_video?

This is the same with bouffalolab official.

marcpabst, 2023-08-22:
I'll give it another try later! I'm compiling under darwin/arm64, do you think that could make a difference? (just spitballing here...)

closed 2023-08-25 by sakumisu