cherry-embedded/CherryUSB #149: USB HS Device : UAC2 issues

OpusElectronics, 2023-11-20:
Hello,

I've been testing CherryUSB on a CH32V307 and it looks great.
Having issues with the audio class though. I've tested the following:

[audio_v2_mic_speaker_multichan_template.c](https://github.com/cherry-embedded/CherryUSB/blob/master/demo/audio_v2_mic_speaker_multichan_template.c)

There is no audio example yet on the following repo: https://github.com/CherryUSB/cherryusb_wch , but following other examples, I have set up a project with the above demo template. The USB device is correctly recognized as a UAC2 device with the corresponding bit format and sample rates defined in the audio_v2_mic_speaker_multichan_template.c code. So, that's encouraging. :)

The part that doesn't work properly though is once we start streaming audio from/to the device. On Linux (tested on 2 different machines, same problem), we get a lot of the following kind of error in dmesg:

```
retire_capture_urb: 18822 callbacks suppressed
```

My guess is that in the usbd_audio_iso_out_callback() and/or usbd_audio_iso_in_callback(), we are not sending/receiving the appropriate number of bytes.

In the demo code, you call:
```usbd_ep_start_read(AUDIO_OUT_EP, read_buffer, AUDIO_OUT_PACKET);``` in usbd_audio_iso_out_callback().
I've tried changing AUDIO_OUT_PACKET to nbytes, but it doesn't change much.
I've also tried adding the following line in usbd_audio_iso_in_callback()
```
usbd_ep_start_write(AUDIO_IN_EP, write_buffer, AUDIO_IN_PACKET);
```
(in your demo code, it's empty), but it doesn't solve the issue either.

There's probably something I've overlooked - could you help with this audio class? That would be great.

sakumisu, 2023-11-20:
Audio does not support on all ch32 chips because with no iso driver( you can ask official for help), thanks.

sakumisu, 2023-11-20:
You can try other chips like hpm or bouffalolab.

OpusElectronics, 2023-11-20:
Thanks. So audio can't work as isochronous transfers are not currently supported in CherryUSB for the CH32 chips?
Is this planned in the future?

sakumisu, 2023-11-20:
> Thanks. So audio can't work as isochronous transfers are not currently supported in CherryUSB for the CH32 chips? Is this planned in the future?

No plan. I do not want to guess iso drivers with official registers without reference.

sakumisu, 2023-11-20:
You can refer to this :https://github.com/nikkov/AudioCardCherry

OpusElectronics, 2023-11-20:
Thanks, I will.

closed 2023-11-21 by sakumisu