cherry-embedded/CherryUSB #151: MSC Writing very slow disk

WGR7, 2023-11-23:
Hi,

I have a very slow media that needs much millisecs to write to disk.

What value should **usbd_msc_sector_write** return if the writing to disk has not completed yet?
If Cherry can delay host to write new data while **usbd_msc_sector_write** is waiting media complete. How could I know if host is asking a write status or sending new data if **usbd_msc_sector_write** was called again by usb interrupt while write is in progress?

Regards.

sakumisu, 2023-11-24:
Hi, will return -ETIMEOUT, default is 5s.

WGR7, 2023-11-24:
So, my function body should be like this?

```
int usbd_msc_sector_write(uint32_t sector, uint8_t *buffer, uint32_t length)
{

    if(GetDiskStatus() == WRITE_IN_PROGRESS){
          return -ETIMEOUT;
     }else{

    disk_write(buffer, sector, length);
    return 0;
    }

}
```

closed 2023-11-24 by WGR7