cherry-embedded/CherryUSB #154: USBH_HUB: Interrupt polling stops after NAK

harbaum, 2023-12-05:
The HUB is not polled if the URB is NAK'd. The following line should also wake the thread up on NAK (when nbytes is 0). So this like should simply be removed:
https://github.com/cherry-embedded/CherryUSB/blob/ce2a438c991b92db35c4ed70be1d85ab3bc68fac/class/hub/usbh_hub.c#L322

Furthermore, i'd suggest adding a few milliseconds delay before
https://github.com/cherry-embedded/CherryUSB/blob/ce2a438c991b92db35c4ed70be1d85ab3bc68fac/class/hub/usbh_hub.c#L629
so the HUB is not polled full throttle.

The value of bInterval should IMHO be used to determine the poll rate.

harbaum, 2023-12-05:
But it happens:

```
static void hub_int_complete_callback(void *arg, int nbytes)
{
    struct usbh_hub *hub = (struct usbh_hub *)arg;

    printf("hub_int_complete_callback(%p, %d)\r\n", arg, nbytes);
    
    if (nbytes > 0) {
        usbh_hub_thread_wakeup(hub);
    }
}
```

results in:

`hub_int_complete_callback(0x62fc9614, 0)`

And the hub stops working. Remove that "if()" and the hub keeps working.

sakumisu, 2023-12-06:
This is driver bug, you should update cherryusb code.

sakumisu, 2023-12-06:
Maybe you use bl616, right? You should update ehci driver with 0.10.2, then you can fix this issue.

harbaum, 2023-12-06:
I've updated to 10.2 and the problem seems to be gone. Thanks a lot!

closed 2023-12-06 by harbaum