cherry-embedded/CherryUSB #155: Wrong max packet size for Mentor USB RXMAP and TXMAP registers

tiancj, 2023-12-06:
In function usbd_ep_open, `MUSB_IND_RXMAP_OFFSET` and/or `MUSB_IND_TXMAP_OFFSET` is set to `ep_cfg->ep_mps`, but according to Mentor USB SW guide, the value of these registers is in unit of 8 Bytes, so the correct value of these registers may be
`ep_cfg->ep_mps >> 3`.

sakumisu, 2023-12-06:
No i think your datasheet is incorrect and not official, please see 
[image]

sakumisu, 2023-12-06:
You can find this config in TI/ES32 or other chips with offical.

tiancj, 2023-12-06:
Thanks, I recheck these two registers, FS and HS have different definitions. FS in 8 bytes, HS in bytes. :-)