cherry-embedded/CherryUSB #157: STM32F407使用USB_OTG_HS的full speed phy做主机枚举不成功

yzl893, 2023-12-26:
STM32F407使用USB_OTG_HS的full speed phy做主机，IO口是PB14\PB15，RTT系统中使用软件包
连接full speed设备均枚举失败，失败的地方不固定，以下几种都会出现
[E/USB] Failed to get device descriptor,errorcode:-14
[E/USB] Failed to set configuration,errorcode:-14
[E/USB] Failed to set devaddr,errorcode:-14
[E/USB] Failed to get full config descriptor,errorcode:-14

手头上有个只支持low speed的键盘接上可以正常枚举和使用，请问可能是什么原因导致的呢？
[image]

sakumisu, 2023-12-27:
电路问题，如果我猜得不错，你是用的杜邦线接到pb14 15的。建议更换板子。

sakumisu, 2023-12-27:
这种情况群里已经好几个了

yzl893, 2023-12-27:
> 电路问题，如果我猜得不错，你是用的杜邦线接到pb14 15的。建议更换板子。

这个现象我也是怀疑电路问题，就找了别的usb host例程来测试，都是可以正常枚举，包括用rtt官方usb驱动也正常，而且板子是pcb走的线，不是杜邦线，应该不是硬件问题，所以就很疑惑。有没有可能是usb_config.h里什么地方参数不对导致的呢？我是复制例程里stm32F429的文件改的，但是按理说low speed能正常，初始化这些应该都没问题才对。

sakumisu, 2023-12-27:
不会，纯电路问题。建议买一个野火板子测试

sakumisu, 2023-12-27:
其他的你也没说，我不清楚你怎么移植的，所以建议使用我提供的demo，多谢。

sakumisu, 2023-12-27:
其余问题，加readme底下群再讨论把

sakumisu, 2023-12-27:
https://github.com/CherryUSB 这边也有rtthread的例程，如果电路没问题，请检查移植，如果不清楚，请直接购买开发板以后，使用我提供的测试。感谢。

yzl893, 2023-12-27:
> https://github.com/CherryUSB/rt-thread 这边也有rtthread的例程，如果电路没问题，请检查移植，如果不清楚移植哪里有问题，请直接购买开发板以后，使用我提供的测试。感谢。

好的，谢谢了，我这边再测试一下看看

sakumisu, 2023-12-30:
问题结束，使用 keil 版本可以使用。

closed 2023-12-30 by sakumisu