cherry-embedded/CherryUSB #158: Later roadmap (after v1.0.0)

sakumisu, 2023-12-30:
- [ ] Support multi port for device
- [ ] Support multi port for host
- [ ] Support adb
- [ ] Support bluetooth host class with zephyr bluetooth stack
- [ ] Nuvoton chips mainly support
- [ ] Phytuim chips mainly support

chenzhihong007, 2024-01-03:
建议msc device支持多lun。

AlexYzhov, 2024-01-06:
感觉设备实现需要一个类似 usb gadget 的分层资源注册-管理框架。读了下源代码，目前CherryUSB无论是资源抽象还是API设计都对多控制器的实现不友好。linux 和 libopencm3 这块设计的不错，但 license 都有传染性。

sakumisu, 2024-01-06:
> 感觉设备实现需要一个类似 usb gadget 的分层资源注册-管理框架。读了下源代码，目前CherryUSB无论是资源抽象还是API设计都对多控制器的实现不友好。linux 和 libopencm3 这块设计的不错，但 license 都有传染性。

不做多控制器支持，不好意思，多port只支持相同ip

AlexYzhov, 2024-01-06:
> > 感觉设备实现需要一个类似 usb gadget 的分层资源注册-管理框架。读了下源代码，目前CherryUSB无论是资源抽象还是API设计都对多控制器的实现不友好。linux 和 libopencm3 这块设计的不错，但 license 都有传染性。
> 
> 不做多控制器支持，不好意思，多port只支持相同ip

哦哦，我这里可能描述的不清晰。前面提到的多控制器支持，是指相同IP多实例的情况

sakumisu, 2024-01-06:
> > > 感觉设备实现需要一个类似 usb gadget 的分层资源注册-管理框架。读了下源代码，目前CherryUSB无论是资源抽象还是API设计都对多控制器的实现不友好。linux 和 libopencm3 这块设计的不错，但 license 都有传染性。
> > 
> > 
> > 不做多控制器支持，不好意思，多port只支持相同ip
> 
> 哦哦，我这里可能描述的不清晰。前面提到的多控制器支持，是指相同IP多实例的情况

这个下次更新就有了，api多个形参的事情

sakumisu, 2024-01-06:
> > > 感觉设备实现需要一个类似 usb gadget 的分层资源注册-管理框架。读了下源代码，目前CherryUSB无论是资源抽象还是API设计都对多控制器的实现不友好。linux 和 libopencm3 这块设计的不错，但 license 都有传染性。
> > 
> > 
> > 不做多控制器支持，不好意思，多port只支持相同ip
> 
> 哦哦，我这里可能描述的不清晰。前面提到的多控制器支持，是指相同IP多实例的情况

不同ip的多port就不做支持了哈，这个就要用户注册驱动，不合适，而且，基本很少芯片，两个port用的不同ip吧

hucya, 2024-03-31:
当我使用STM32G4的时候 CONFIG_USBDEV_FSDEV_PMA_ACCESS 这个宏定义 为什么要是1才能正常运行，参考STM32F1系列的，配置的默认2，一直运行不了。

sakumisu, 2024-03-31:
> 当我使用STM32G4的时候 CONFIG_USBDEV_FSDEV_PMA_ACCESS 这个宏定义 为什么要是1才能正常运行，参考STM32F1系列的，配置的默认2，一直运行不了。

里面已经说了，要填1或者2，具体多少需要根据芯片厂家定义修改，g4应该参考st g4的驱动，而不是f1

cdmar79, 2024-12-25:
as 'demo' folder as so many templete, would you please make a list to describe the functionarity and usage of these all.

sakumisu, 2024-12-25:
You can get functionality‌ from its name and doc. For usage, it is useless to describe for demo and we provide every file usage about cherryusb, demo is just a application or a template and they are variable by users, so its usage is useless.

closed 2025-01-01 by sakumisu