cherry-embedded/CherryUSB #164: PMA_ACCESS

yffbit, 2024-01-21:
这两处为什么有的乘上PMA_ACCESS有的没乘
https://github.com/cherry-embedded/CherryUSB/blob/04a300d80306fa57e412044d52a00876a9ce81ce/port/fsdev/usb_fsdev_reg.h#L1476-L1477
https://github.com/cherry-embedded/CherryUSB/blob/04a300d80306fa57e412044d52a00876a9ce81ce/port/fsdev/usb_fsdev_reg.h#L1446-L1447
附上`STM32_USB-FS-Device_Lib_V4.1.0\Libraries\STM32_USB-FS-Device_Driver\inc\usb_regs.h`中的相关代码
[image]

sakumisu, 2024-01-21:
这个你应该去问ST，这是从ST那复制的

sakumisu, 2024-01-21:
请使用官方标准的hal库对比，而不是你这种已经废弃的旧库

closed 2024-01-21 by sakumisu

sakumisu, 2024-01-22:
https://github.com/CherryUSB/cherryusb_stm32/blob/master/usb_device/stm32f103c8t6/Drivers/STM32F1xx_HAL_Driver/Inc/stm32f1xx_hal_pcd.h

sakumisu, 2024-01-22:
总之，×不乘 PMA_ACCESS 是 IP 设计人的行为，不同芯片，想改就改，不是我们软件的控制

sakumisu, 2024-01-22:
你这只能说明F3 PMA_ACCESS =1 罢了，这很正常，大多数是=2，少数=1。觉得不统一，就去喷设计 IP 的就行了

yffbit, 2024-01-22:
我的意思是这4行代码是统一的，要么都不乘，要么都乘2
你这样写，如果定义`PMA_ACCESS = 2`，这4行代码就不统一了
另外，有些mcu的HAL库里面根本没有定义`PCD_EP_TX_ADDRESS`
https://github.com/search?q=PCD_EP_TX_ADDRESS&type=code
我猜你应该是把不同mcu的定义放在一起了

sakumisu, 2024-01-22:
我上面已经发了链接了，这是st设计，而且这个宏没有用到，请参考PCD_SET_EP开头，而不是PCD_EP开头

sakumisu, 2024-01-22:
所以你说的这些宏，全都可以忽略，因为压根就没用到

sakumisu, 2024-01-22:
为防止误解，已经删除

sakumisu, 2024-01-22:
> 我的意思是这4行代码是统一的，要么都不乘，要么都乘2 你这样写，如果定义`PMA_ACCESS = 2`，这4行代码就不统一了 另外，有些mcu的HAL库里面根本没有定义`PCD_EP_TX_ADDRESS` https://github.com/search?q=PCD_EP_TX_ADDRESS&type=code 我猜你应该是把不同mcu的定义放在一起了

应该是因为官方没有定义，所以这边复制的时候没考虑，但是这个也没用，所以没有影响。建议让官方增加