cherry-embedded/CherryUSB #165: Please provide a Bouffalo SDK host example

harbaum, 2024-01-22:
The cherryusb inside the bouffalo sdk is still 0.10.0. I have been able to manually update to 0.10.2 but the latest version 1.0.0 doesn't implement easily into the bouffalo SDK, anymore.

Please provide an example for USB (host) usage of the latest cherryusb 1.0.0 with the bouffalo SDK. Or even better, please update the bouffalo SDK with the latest cherryusb.

sakumisu, 2024-01-22:
Sorry, not updatevany more

sakumisu, 2024-01-22:
You can update by yourself with guide.

harbaum, 2024-01-22:
I don't understand. The bouffalo SDK will not be updated, anymore? Or only the USB inside the bouffalo SDK? 

What do you mean by guide? I tried updating. Main changes are:

- removing `usbd_ep_open()` from `bflb_usb_v2.c` 
- pointing to `extern struct usb_setup_packet g_setup_buffer[];`  in `bflb_usb_v2.c `
- and replaced` usbh_initialize() ` by  `usbh_initialize(usbh_alloc_bus(0, BLFB_USB_BASE))`

I had to do some minor other changes in my code. But I had already done this for 0.10.2 and that was still working

This is now crashing once I plug a USB device with:

```
init usbh class
exception_entry
mcause=30000004
mepc:a00064d8
mtval:0000002d
Load address misaligned

-+-+-+- BFLB COREDUMP v0.0.1 +-+-+-+
...
```

sakumisu, 2024-01-22:
Follow this guide: https://cherryusb.readthedocs.io/zh-cn/latest/quick_start/other_chip.html, I think you have wrong in porting with cherryusb master, so please use v0.10.2

sakumisu, 2024-01-22:
From your code, you have many mistakes
- do not remove usbd_ep_open()
- pointing to extern struct usb_setup_packet g_setup_buffer[]; in bflb_usb_v2.c  what is this?
- and replacedusbh_initialize() by usbh_initialize(usbh_alloc_bus(0, BLFB_USB_BASE)), only this cannot work.

sakumisu, 2024-01-22:
For host, where is your irq??? please refer to https://github.com/CherryUSB/cherryusb_stm32/blob/5727f65f5a3da97698b2581cc3055258a57ed182/usb_host/stm32f429igt6/Core/Src/stm32f4xx_it.c#L216

sakumisu, 2024-01-22:
And ehci offset you muse set.

sakumisu, 2024-01-22:
So, following above, please use v0.10.2 if you have no ability to proting with master.

sakumisu, 2024-01-22:
Or you can use v1.0.0, its api is the same with v0.10.2, master is for multi port.

sakumisu, 2024-01-22:
For device, please refer to this commit : https://github.com/cherry-embedded/CherryUSB/commit/29d45ef5c7547b75a347e992fc4198f03da31f15

sakumisu, 2024-01-22:
Follow this #166

harbaum, 2024-01-22:
Yes,  usbd_ep_open() needs to be present. The fact that it compiles without it means that all invocations of this functions are optimized away which in turn means that some quite important call is missing somewhere when doing this naived simple way of just replacing the old components/usb/cherryusb with the updated one.

It's rather strange to hear that support for the bouffalo line of MCUs has ended so early. Why is that?

harbaum, 2024-01-22:
> Follow this #166

Thanks! This was the info I was missing. USB host compiles now and seems to run.

closed 2024-01-23 by sakumisu

sakumisu, 2024-01-23:
Hello, do you have solved this？For detail, you can check with following patches.

- change `const struct usbd_endpoint_cfg *ep_cfg` to `const struct usb_endpoint_descriptor *ep` or revert change.
- Follow #166  and modify with `bflb_irq_attach(37, USBH_IRQ, NULL);` in bflb_usb_v2.c

harbaum, 2024-01-24:
Yes, I have integrated cherryusb 1.0.0 into the bouffalo sdk. I have also added a host driver for the Xbox controller and compatible devices. See http://GitHub.com/Harbaum/bouffalo_sdk

This is being used in the MiSTeryNano FPGA: http://GitHub.com/Harbaum/MiSTeryNano

sakumisu, 2024-01-25:
> Yes, I have integrated cherryusb 1.0.0 into the bouffalo sdk. I have also added a host driver for the Xbox controller and compatible devices. See http://GitHub.com/Harbaum/bouffalo_sdk
> 
> This is being used in the MiSTeryNano FPGA: http://GitHub.com/Harbaum/MiSTeryNano

That's cool, sir, i will be continuous concerned with your project.