cherry-embedded/CherryUSB #168: 1 RNDIS + 3 CDC Endpoints example demo

sidd-kishan, 2024-01-26:
I am developing a Pi PICO W based RNDIS USB WiFi dongle for windows on ARM platform.
I have needed an example with RNDIS and multiple serial EPs.

RNDIS - For internet
CDC EP2 - SSID SCAN
CDC EP3 - Credential management
CDC EP4 - MISC

I have created a template example : [sidd-kishan/pico-examples-cherryusb](https://github.com/sidd-kishan/pico-examples-cherryusb/blob/3018e100751c1b2f9336846268f886f42bb9e95a/cherryusb/device/rndis_http_server/src/cdc_rndis_device.c)

Please let me know if the same can be supported as a formal demo example

sakumisu, 2024-01-28:
You should confirm if rp2040 has so much endpoints

sakumisu, 2024-01-28:
Your code is right if it supports these endpoints.

sidd-kishan, 2024-01-29:
Hi Team,
              PFB the screenshot from the latest build from my project [sidd-kishan/pico-examples-cherryusb](https://github.com/sidd-kishan/pico-examples-cherryusb) Please let me know if it is sufficient as an evidence:-

[image]

sakumisu, 2024-01-29:
It works ok.

sidd-kishan, 2024-01-29:
Updated the code now all 3 endpoints working as [independent loopback](https://github.com/sidd-kishan/pico-examples-cherryusb/blob/master/cherryusb/device/rndis_http_server/src/cdc_rndis_device.c)

closed 2024-01-30 by sakumisu