cherry-embedded/CherryUSB #183: Cherry USB host wifi and more ... on Teensy 4.x

JPB1JPB1, 2024-03-14:
Dear Team

any example to get the Cherry USB host wifi, bluetooth, pendrive stack to run with Teensy 4.1 ?
Few applications may use your stack..
Waiting for some support and example to test it asap.
Best Regards
Jjj

sakumisu, 2024-03-14:
I dont know what is teensy, you can run bluetooth host with zephyr bluetooth stack, no support for host wifi because it need too much effort.

JPB1JPB1, 2024-03-14:
Dear Team

here are the Teensy 4.x board https://www.pjrc.com/store/teensy41.html
Cortex M7 at 600Mhz , USB host High speed, low power... 
What about RT-Thread also ?
Best Regards
Jjj

sakumisu, 2024-03-14:
I see it uses ehci ip, but this should work by yourself.You can use other support chips with ehci ip like bl616 or hpm or nuvoton.

JPB1JPB1, 2024-03-14:
Dear Sakumisu
Ok, please, informations are welcome to help me to support any other USB dongle ...
May be in private message to discuss how to do it...
Best regards
Jjj

closed 2024-03-15 by sakumisu