cherry-embedded/CherryUSB #195: How to implement USB host class driver that supports many vid/pid pairs?

harbaum, 2024-04-22:
I've implemented a USB host class driver for xbox controllers here: 
  https://github.com/harbaum/CherryUSB/tree/e0683fc13d199d9ef49912c8e08bcd5dcf59ea20/class/xbox

This needs support for a long list of vid/pid pairs like e.g. 
https://github.com/harbaum/CherryUSB/blob/e0683fc13d199d9ef49912c8e08bcd5dcf59ea20/class/xbox/usbh_xbox.c#L111-L215

To achieve this I have changed the single vid/pid entry to point to an array of entries like so: https://github.com/harbaum/CherryUSB/blob/e0683fc13d199d9ef49912c8e08bcd5dcf59ea20/core/usbh_core.h#L61

This is imcompatible with upstream cherryusb.

How do I properly implement a driver that supports multiple vid/pid pairs in cherryusb?

sakumisu, 2024-04-22:
Thanks, i will support id table for you.

harbaum, 2024-04-22:
I've just updated my port to the latest bouffalolab_sdk and the cherryusb 1.2,0. Feel free to re-use any part you want.

I've also updated https://github.com/harbaum/bouffalo_sdk/blob/master/drivers/lhal/src/bflb_usb_v2.c for the latest cherryusb 1.2.0

sakumisu, 2024-04-23:
Thank you sir, i have reviewed your code, and give you a few suggestions:
- in `usbh_find_class_driver` , if any flag does not match, please continue instead of going on next.
- some changes in vendor do not modify
- change `USB_CLASS_MATCH_DEVICE_LIST` with `USB_CLASS_MATCH_ID`
- format your code.

And after doing all, you can create a pr, sincerely wait and thank you for your pr.

harbaum, 2024-04-23:
Solved in a local branch. Thanks

closed 2024-04-23 by harbaum

sakumisu, 2024-05-04:
Add this feature refer to your code, thank you.