cherry-embedded/CherryUSB #197: Can't

bhxxwx, 2024-04-28:


bhxxwx, 2024-04-28:
Hi, I transplanted the program to the ESP32-S3 chip, and an error occurred when running: 
`[E/USB] This dwc2 version does not support dma, so stop working`
Above this message, I found that the information of dwc2 hcd could not be obtained normally,  output below:
```
[I/USB] ========== dwc2 hcd params ==========
[I/USB] CID:00000000
[I/USB] GSNPSID:00000000
[I/USB] GHWCFG1:00000000
[I/USB] GHWCFG2:00000000
[I/USB] GHWCFG3:00000000
[I/USB] GHWCFG4:00000000
[I/USB] dwc2 has 1 channels
```
My esp32-S3 chip version is V0.2, build in ESP-IDF V5.2.1
I'm wondering if the ESP changes the address of the USB peripheral?
In this program:
`#define USB_BASE 0x60080000`

sakumisu, 2024-04-28:
Your usb low level init is incorrect, thanks.

sakumisu, 2024-04-28:
Do you think all of usb registers are all zero? No.

sakumisu, 2024-04-28:
Refer to https://github.com/CherryUSB/cherryusb_esp32/tree/latest

closed 2024-04-28 by sakumisu

bhxxwx, 2024-04-29:
Thank you for your reminder. The `usb_glue_esp.c` file is missing from the component I downloaded. Now the program can recognize the USB insertion, but fails to obtain the descriptor, prompting -116 which is TimeOut, how can i fix this?
thank you very much again.

```
[I/USB] cherryusb, version: 0x001001
[I/USB] ========== dwc2 hcd params ==========
[I/USB] CID:00000000
[I/USB] GSNPSID:4f54400a
[I/USB] GHWCFG1:00000000
[I/USB] GHWCFG2:224dd930
[I/USB] GHWCFG3:00c804b5
[I/USB] GHWCFG4:d3f0a030
[I/USB] dwc2 has 8 channels
[I/USB] New full-speed device on Hub 1, Port 1 connected
[E/USB] Failed to get device descriptor,errorcode:-116
[E/USB] Port 1 enumerate fail
```

sakumisu, 2024-04-29:
I have said, use my demo repo, thanks.

sakumisu, 2024-04-29:
-116 is very normal because of incorrect dwc2 config.

sakumisu, 2024-04-29:
If you do not use my demo repo, do not reply for this.

bhxxwx, 2024-04-29:
ok, thanks again

sakumisu, 2024-04-29:
Default dwc2 config need 4K fifo, but esp32 only has 1K, so it cannot work.Do not use old version.