cherry-embedded/CherryUSB #205: BL616 USB MSC Long Delay Write Issue

WGR7, 2024-05-19:
Hi,

I'm using a BL616 with USB HS mode. I've updated component to more recent master CherryUSB branch.

I have a very slow media: a NOR Flash SPI. I'm getting a issue while writing to the media. In the middle of transfer I get a type of reconfig of interfaces with these messages. They are the same message when the USB cable is plugged in USB port. The file to write is large, about 40MBytes.

> [D/USB] write lba:152
> [D/USB] EP0 send 18 bytes, 0 remained
> [D/USB] EP0 send 18 bytes, 0 remained
> [D/USB] EP0 send 64 bytes, 34 remained
> [D/USB] EP0 send 34 bytes, 0 remained
> [D/USB] EP0 send 22 bytes, 0 remained
> [I/USB] Open ep:0x86 type:3 mps:8
> [I/USB] Open ep:0x02 type:2 mps:512
> [I/USB] Open ep:0x81 type:2 mps:512
> [I/USB] Open ep:0x04 type:2 mps:512
> [I/USB] Open ep:0x83 type:2 mps:512

To isolate the problem I did take the example of the composite device cdc_mscram  and in the final of the write function I've added a vTaksDelay(200).

```
int usbd_msc_sector_write(uint8_t busid, uint8_t lun, uint32_t sector, uint8_t *buffer, uint32_t length)
{
    if (sector < 10)
        memcpy(mass_block[sector].BlockSpace, buffer, length);

    vTaskDelay(200);
    return 0;
}
```

The issue is still the same, the device is reconfigured and the transfer stops.

> [D/USB] 2
> [D/USB] read lba:196
> [D/USB] 2
> [D/USB] read lba:197
> [D/USB] 2
> [D/USB] read lba:198
> [D/USB] 2
> [D/USB] read lba:199
> [D/USB] Send csw
> [D/USB] Start reading cbw
> [D/USB] Decode CB:0x28
> [D/USB] lba: 0x0100
> [D/USB] nsectors: 0x08
> [D/USB] 2
> [D/USB] read lba:256
> [D/USB] 2
> [D/USB] read lba:257
> [D/USB] 2
> [D/USB] read lba:258
> [D/USB] 2
> [D/USB] read lba:259
> [D/USB] 2
> [D/USB] read lba:260
> [D/USB] 2
> [D/USB] read lba:261
> [D/USB] 2
> [D/USB] read lba:262
> [D/USB] 2
> [D/USB] read lba:263
> [D/USB] Send csw
> [D/USB] Start reading cbw
> [D/USB] Decode CB:0x2a
> [D/USB] lba: 0x0000
> [D/USB] nsectors: 0xc8
> [D/USB] 1
> [D/USB] write lba:0
> [D/USB] 1
[... 639 lines cut ...]
> [D/USB] 1
> [D/USB] write lba:151
> [D/USB] 1
> [D/USB] write lba:152
> [D/USB] EP0 send 18 bytes, 0 remained
> [D/USB] EP0 send 18 bytes, 0 remained
> [D/USB] EP0 send 64 bytes, 34 remained
> [D/USB] EP0 send 34 bytes, 0 remained
> [D/USB] EP0 send 22 bytes, 0 remained
> [I/USB] Open ep:0x86 type:3 mps:8
> [I/USB] Open ep:0x02 type:2 mps:512
> [I/USB] Open ep:0x81 type:2 mps:512
> [I/USB] Open ep:0x04 type:2 mps:512
> [I/USB] Open ep:0x83 type:2 mps:512
> [D/USB] Start reading cbw
> [D/USB] Start reading 7 bytes from ep0
> [D/USB] EP0 recv 7 bytes, 0 remained
> [D/USB] CDC Class request: bRequest 0x20
> [D/USB] Set intf:0 linecoding <9600 8 N 1>
> [D/USB] MSC Class request: bRequest 0xff
> [E/USB] ep:83 clear halt
> [E/USB] ep:04 clear halt
> [D/USB] Decode CB:0x00
> [D/USB] Send csw
> [D/USB] Start reading cbw

sakumisu, 2024-05-19:
Host do not dupport too slow msc, so stop useing it, and your code is incorrect because  write api is in isr.

WGR7, 2024-05-20:
BL616 is used as a MSC device, insn't a host. MSC is a blocking protocol as I know, so the write can be slow. I'm using FreeRTOS.

sakumisu, 2024-05-20:
> BL616 is used as a MSC device, insn't a host. MSC is a blocking protocol as I know, so the write can be slow. I'm using FreeRTOS.

i hace said, cherryusb write api is in isr,and host is pc.

sakumisu, 2024-05-20:
usbd_msc_sector_write is in isr thanks.Please give up if your flash is too slow, it is not cherryusb problem.

closed 2024-05-20 by sakumisu

WGR7, 2024-05-20:
I'll take my time and port TinyUSB. your library is a waste of time. Tks.

sakumisu, 2024-05-20:
So jokes, you should study freertos first and know freertos api cannot be used in interrupt api

WGR7, 2024-05-20:
I can patch your code to deal with this issue, but I need to know how can I do NAKs to host when the MCU is writing to slow memory. Is there any specific SCSI status to return that can I inform host that device is busy?