cherry-embedded/CherryUSB #220: uint8_t class; in CherryUSB conflict with reserved keyword c++

obitvn, 2024-07-11:
[image]
Hi, I am trying to use cherryUSB on ESP32 with C++. I had it running on previous versions, but in the current version, the usbh_class_info structure in usbh_core.h contains the keyword class, causing a conflict with C++.
[image]
So we should consider changing class to another keyword (I have seen this issue since 2022, in issues #8 , but it seems it has not been completely fixed)?

obitvn, 2024-07-11:
I'm also trying to find a solution that doesn't modify the CherryUSB source code, but it seems extern "C" { } doesn't help.

sakumisu, 2024-07-11:
This #8 is fixed for device and no host stack at that time, i will consider to fix this in next release.

sakumisu, 2024-07-11:
You can follow this demo https://github.com/CherryUSB/cherryusb_esp32/tree/latest, and do not include usbh_core.h in c++ source file.

obitvn, 2024-07-11:
Thank you, this works.

closed 2024-07-11 by obitvn