cherry-embedded/CherryUSB #222: BL616 reports "port 1 does not enable power"

harbaum, 2024-07-15:
I just updated my bl616 setup to the latest cherryusb 1.3.1 and since then the bl616 runs into this error:

```
[I/USB] EHCI HCIVERSION:0x0100
[I/USB] EHCI HCSPARAMS:0x000001
[I/USB] EHCI HCCPARAMS:0x0006
[I/USB] EHCI ppc:0, n_ports:1, n_cc:0, n_pcc:0
[I/USB] EHCI uses tt for ls/fs device
[W/usbh_hub] Port 1 does not enable power
```

The attached device is not accessed after that. Still unplugging is detected:

`[I/usbh_hub] Device on Bus 0, Hub 1, Port 1 disconnected`

Replugging it gives the same problem:
`
[W/usbh_hub] Port 1 does not enable power`

sakumisu, 2024-07-15:
You should checkout v1.3.1 not master

sakumisu, 2024-07-15:
[image]

sakumisu, 2024-07-15:
So this is the bug of bl616, it means bl616 ehci ip is not standard

closed 2024-07-15 by sakumisu

harbaum, 2024-07-15:
Changed like so:

```
#if defined(BL616)
    g_ehci_hcd[bus->hcd.hcd_id].ppc = true;
#else
    g_ehci_hcd[bus->hcd.hcd_id].ppc = (EHCI_HCCR->hcsparams & EHCI_HCSPARAMS_PPC) ? true : false;
#endif
```

The result has ppc set. But that does not solved the problem:

```
[I/USB] EHCI HCIVERSION:0x0100
[I/USB] EHCI HCSPARAMS:0x000001
[I/USB] EHCI HCCPARAMS:0x0006
[I/USB] EHCI ppc:1, n_ports:1, n_cc:0, n_pcc:0
[I/USB] EHCI uses tt for ls/fs device
[W/usbh_hub] Port 1 does not enable power
```

sakumisu, 2024-07-15:
remove if in

```
                if (temp & EHCI_PORTSC_PP) {
                    status |= (1 << HUB_PORT_FEATURE_POWER);
                }
```

sakumisu, 2024-07-15:
Because bl616 ehci does not have ppc, so temp will not match EHCI_PORTSC_PP

harbaum, 2024-07-15:
That fixes it! Thanks. 

```
#if !defined(BL616)
                if (temp & EHCI_PORTSC_PP)
#endif
	        {
                    status |= (1 << HUB_PORT_FEATURE_POWER);
                }
```

sakumisu, 2024-07-15:
I hate those off-standard ips because we need to do many patches for them.

harbaum, 2024-07-15:
What about:

```
                if (temp & EHCI_PORTSC_PP || !(EHCI_HCCR->hcsparams & EHCI_HCSPARAMS_PPC) ) {
                    status |= (1 << HUB_PORT_FEATURE_POWER);
                }

```

?

sakumisu, 2024-07-15:
Just always enable this bit i think.

harbaum, 2024-07-15:
The code above doesn't change your code logic on other platforms but fixes the bl616 without and ifdef. Works for me ...

sakumisu, 2024-07-15:
You are right, can you push mr for this patch?

sakumisu, 2024-07-15:
This feature is not in v1.3.1 but in master branch, for stable, you can checkout release tag.