cherry-embedded/CherryUSB #225: 指针赋值错误导致property无法正确获取

KeroChen, 2024-07-15:
问题描述：使用demo中的winusb1.0例程，测试WinUsb功能时，在上位机中使用正确的GUID一直无法匹配到目标USB设备。
产生原因：在usbd_core.c的usbd_vendor_request_handler处理函数中，给data指针赋值comp_id_property的地址操作不正确。

例如
原始代码：
```
*data = (uint8_t *)g_usbd_core[busid].msosv1_desc->comp_id_property[setup->wValue];
```
应改为：
```
*data = (uint8_t *)&g_usbd_core[busid].msosv1_desc->comp_id_property[setup->wValue];
```

代码修改后，在设备管理器中卸载设备，使用上位机再次测试可以成功匹配到目标USB设备。

同时也发现了其他地方也有类似指针赋值不正确的操作，需要仔细审查。

sakumisu, 2024-07-15:
这边没又问题，over，这是二级指针

closed 2024-07-15 by sakumisu

sakumisu, 2024-07-15:
demo 中修改成如下即可
```
const uint8_t *WINUSB_IFx_WCIDProperties[] = {
#if DOUBLE_WINUSB == 0
    WINUSB_IF0_WCIDProperties,
#else
    WINUSB_IF1_WCIDProperties,
#endif
};

struct usb_msosv1_descriptor msosv1_desc = {
    .string = WCID_StringDescriptor_MSOS,
    .vendor_code = WCID_VENDOR_CODE,
    .compat_id = WINUSB_WCIDDescriptor,
    .comp_id_property = WINUSB_IFx_WCIDProperties,
};

```

KeroChen, 2024-07-15:
噢，demo的问题哈？明白了

sakumisu, 2024-07-15:
晚点提交patch，比较懒，更新不及时

sakumisu, 2024-07-15:
重新更正
```
const uint8_t *WINUSB_IFx_WCIDProperties[] = {
    WINUSB_IF0_WCIDProperties,
#if DOUBLE_WINUSB == 1
    WINUSB_IF1_WCIDProperties,
#endif
};
```