cherry-embedded/CherryUSB #226: Version in project specific file?

harbaum, 2024-07-15:
What's the idea behind putting the version and version string into the usb_config.h ? 

These are imho to be handled in the projects and thus won't reflect the state of cherryusb.

Can the version please be moved into the charryusb main files? I can then do things like `#if CHERRYUSB_VERSION > 0x010300` and the like.

sakumisu, 2024-07-15:
You can include xxx_core.h and it inculdes usb_config.h. The reason why version is in usb_config is that anyone who wants to ask some questions and they do not use git, they should provide version and then i can know the changes, this version in usb_config is readonly.

harbaum, 2024-07-15:
Yes. But a project like mine comes with it's own project specific usb_config.h and that won't necessarily match the cherryusb stack. E.g. here: https://github.com/harbaum/FPGA-Companion/blob/main/src/bl616/usb_config.h

harbaum, 2024-07-15:
Ok, will do that.

closed 2024-07-15 by harbaum

sakumisu, 2024-08-16:
Moved macros into usb_version.h