cherry-embedded/CherryUSB #227: AT32F405xx 在macos下  CONFIGURED 后会收到 SET_REMOTE_WAKEUP

Clearhu, 2024-07-22:
AT32F405xx 开启HS，以及DMA，在Macos下  CONFIGURED 后会收到 SET_REMOTE_WAKEUP
Windows 下没有该问题。

sakumisu, 2024-07-22:
忽略即可

Clearhu, 2024-07-22:
> 忽略即可

如果忽略，会影响正常的suspend，还有其他方法吗？

sakumisu, 2024-07-22:
描述符中关闭 remote wakeup功能，其次，如果使用remote 功能，需要自行实现所有wakeup相关的内容，包括低功耗，suspend处理。你如果不需要的话，suspend无所谓我觉得

Clearhu, 2024-07-22:
> 描述符中关闭 remote wakeup功能，其次，如果使用remote 功能，需要自行实现所有wakeup相关的内容，包括低功耗，suspend处理。你如果不需要的话，suspend无所谓我觉得

现在确实需要使用 remote wakeup，我感觉奇怪的是，为什么win上是正常的，是不是在 usbd_std_device_req_handler的USB_REQUEST_SET_FEATURE这里判断一下条件来决定正常的suspend

sakumisu, 2024-07-22:
如果你win休眠后没发remote wakeup，不应该是不正常吗。其次，不需要判断什么条件，USB_REQUEST_SET_FEATURE 只是对主机发送的命令进行接收并且提供回调 callback，至于回调里面需要做什么低功耗，suspend处理，这些是用户自行处理，协议栈只负责处理命令和回调

Clearhu, 2024-07-22:
> 如果你win休眠后没发remote wakeup，不应该是不正常吗。其次，不需要判断什么条件，USB_REQUEST_SET_FEATURE 只是对主机发送的命令进行接收并且提供回调 callback，至于回调里面需要做什么低功耗，suspend处理，这些是用户自行处理，协议栈只负责处理命令和回调

可能我描述的不清楚，现在的问题是，上电后在macos下usb收到 USBD_EVENT_CONFIGURE 后马上会收到一条 USBD_EVENT_SET_REMOTE_WAKEUP，这样就导致产品进入了suspend逻辑。

sakumisu, 2024-07-22:
那是你代码问题，我们正常处理的时候是不会在 set remote 里面做suspend处理，而是等remote wakeup和 suspend callback同时触发才会做suspend

Clearhu, 2024-07-22:
> 那是你代码问题，我们正常处理的时候是不会在 set remote 里面做suspend处理，而是等remote wakeup和 suspend callback同时满足才会做suspend

谢谢，那我明白了，只是在usb_glue_at.c 里面并没有调用 usbd_event_suspend_handler，是需要自己实现是对吗？

sakumisu, 2024-07-22:
这是 ip层面的事情，ip里面没有做

sakumisu, 2024-07-22:
目前仅测试 过bl，hpm支持

Clearhu, 2024-07-22:
> 这是 ip层面的事情，ip里面没有做

好的，谢谢，可以关闭issues.
感谢！！

closed 2024-07-22 by Clearhu