cherry-embedded/CherryUSB #229: Uf2 bootloader for BL616

marchingband, 2024-08-04:
How can I port the UF2 bootloader to BL616 MCU?
Is it primarily just the read/write flash functions that need to be completed?
I am looking at the template file, but it is not obvious to me how to proceed.
thank you!

sakumisu, 2024-08-05:
You can implement bootuf2_flash_write api to write your firmware and use bootuf2_is_write_done to check if firmware is writing done and then reboot or jump.

marchingband, 2024-08-05:
The API does not match the version of CherryUsb which the Bouffalo SDK contains.
Looking through the tagged versions of CherryUsb, I notice that demo/bootuf2/ does not exist in the previous releases.
Will it be possible to port? Any tips?

sakumisu, 2024-08-05:
You can copy bootuf2 to your old cherryusb ,that's easy to port.

marchingband, 2024-08-05:
It was easy to port, but I do not see an example of updating code OTA, or using partition table in the BL SDK.
There are some examples in the bluetooth examples folder, such as `oad_client` but they are not documented well enough for me to follow.

sakumisu, 2024-08-05:
This is example for all mcu not only for one chip like bl.