cherry-embedded/CherryUSB #232: 移植到CH57X上使用cdc_acm_template.c,无法成功进行USB通信

wdfk-prog, 2024-08-05:
### 硬件是否正常，是否使用杜邦线连接，如果正常，请说明正常原因
- 硬件正常,使用沁恒提供demo可以正常创建虚拟串口

### USB 中断，时钟，引脚，寄存器地址是否正确，截图
- 正常,该芯片寄存器地址与CH58X一致
- 直接使用的是usb_ch58x_dc_usbfs.c作为port
- 可以看到完全一致
[image]

### 是否能进 USB 中断
- 可以进入,中断中添加日志可以输出
[image]

### 芯片是否带有 cache功能，是否做了 no cache 处理，截图
- 无cache

### 打开 CONFGI_USBDEV_SETUP_LOG_PRINT，并提供 log

- 中断中添加打印如下

```c
/**
 * @brief            USB interrupt processing function
 * @pre              None
 * @param[in]        None
 * @retval           None
 */
__attribute__((interrupt("WCH-Interrupt-fast")))
__attribute__((section(".highcode"))) void
USBD_IRQHandler(void)
{
    volatile uint8_t intflag = 0;
    intflag = CH58x_USBFS_DEV->USB_INT_FG;
    if (intflag & RB_UIF_TRANSFER) {
        printf("RB_UIF_TRANSFER: %x\r\n", CH58x_USBFS_DEV->USB_INT_ST);
        if ((CH58x_USBFS_DEV->USB_INT_ST & MASK_UIS_TOKEN) != MASK_UIS_TOKEN) {
            printf("RB_UIF_TRANSFER: %x\r\n", CH58x_USBFS_DEV->USB_INT_ST);
            uint8_t epid = ((CH58x_USBFS_DEV->USB_INT_ST & (MASK_UIS_TOKEN | MASK_UIS_ENDP)) & 0x0f);
            printf("epid: 0x%x\r\n", epid);
            printf("UIS_TOKEN %x\r\n", (CH58x_USBFS_DEV->USB_INT_ST & (MASK_UIS_TOKEN | MASK_UIS_ENDP)) & 0xf0);
            switch ((CH58x_USBFS_DEV->USB_INT_ST & (MASK_UIS_TOKEN | MASK_UIS_ENDP)) & 0xf0) {
                case UIS_TOKEN_IN:
                    if (epid == 0) {
                        /**
                         * IN  The host takes away the data that has been stored in FIFO
                         */
                        printf("FIFO: 0x%x\r\n", usb_dc_cfg.setup.bmRequestType >> USB_REQUEST_DIR_SHIFT);
                        switch (usb_dc_cfg.setup.bmRequestType >> USB_REQUEST_DIR_SHIFT) {
                            case 1:
                                /*!< Get */
                                CH58x_USBFS_DEV->UEP0_CTRL ^= RB_UEP_T_TOG;
                                /**
                                 * Here is to take away the last data, and the IN interrupt will be triggered only after it is successfully taken away.
                                 * Therefore, the status of the in endpoint is set to NAK here. If there is data transmission,
                                 * the endpoint status will be set to ack again in the in handler of EP0.
                                 */
                                EPn_SET_TX_NAK(0);

                                /*!< IN */
                                if (usb_dc_cfg.ep_in[0].xfer_len > usb_dc_cfg.ep_in[0].mps) {
                                    usb_dc_cfg.ep_in[0].xfer_len -= usb_dc_cfg.ep_in[0].mps;
                                    usb_dc_cfg.ep_in[0].actual_xfer_len += usb_dc_cfg.ep_in[0].mps;
                                    usbd_event_ep_in_complete_handler(0, 0 | 0x80, usb_dc_cfg.ep_in[0].actual_xfer_len);
                                } else {
                                    usb_dc_cfg.ep_in[0].actual_xfer_len += usb_dc_cfg.ep_in[0].xfer_len;
                                    usb_dc_cfg.ep_in[0].xfer_len = 0;
                                    usbd_event_ep_in_complete_handler(0, 0 | 0x80, usb_dc_cfg.ep_in[0].actual_xfer_len);
                                }
                                break;
                            case 0:
[... 179 lines cut ...]
UIS_TOKEN 20
FIFO: 0x0
RB_UIF_TRANSFER: 40
RB_UIF_TRANSFER: 40
epid: 0x0
UIS_TOKEN 0
usbd_event_ep_out_complete_handler: busid=0, ep=0, nbytes=0
[D/USB] EP0 recv out status
RB_UIF_TRANSFER: f0
RB_UIS_SETUP_ACT: f0
RB_UIF_TRANSFER: 60
RB_UIF_TRANSFER: 60
epid: 0x0
UIS_TOKEN 20
FIFO: 0x0
RB_UIF_TRANSFER: 40
RB_UIF_TRANSFER: 40
epid: 0x0
UIS_TOKEN 0
usbd_event_ep_out_complete_handler: busid=0, ep=0, nbytes=0
[D/USB] EP0 recv out status
```

### 是否流片并销售 
是

sakumisu, 2024-08-05:
1,禁止中断中打印，2，ch不做支持，自行解决

closed 2024-08-05 by sakumisu

wdfk-prog, 2024-08-05:
> 1,禁止中断中打印，2，ch不做支持，自行解决

^_^,好吧

sakumisu, 2024-08-05:
最后，中断里加打印，影响usb时序，就不要想着能用了

wdfk-prog, 2024-08-06:
- 还是多谢了;我多琢磨一下