cherry-embedded/CherryUSB #237: [dwc2][dc]dwc2_set_mode是否需要添加延时确保生效?

wdfk-prog, 2024-08-13:
https://github.com/cherry-embedded/CherryUSB/blob/70944027c50375cda040587980fa2e92e985f536/port/dwc2/usb_dc_dwc2.c#L170-L179

- 观察STM32H7手册发现OTG_GUSBCFG->FDMOD的描述发现`将强制位置 1 后，应用程序必须等待至少 25 ms 后更改方可生效。`
[image]

- 观察STM32驱动代码也是如此
https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/b2b4c45001f1b6b72ae449713607896fd6ca4c6b/Src/stm32h7xx_ll_usb.c#L244-L290

sakumisu, 2024-08-13:
应该是需要的

wdfk-prog, 2024-08-13:
https://github.com/cherry-embedded/CherryUSB/blob/70944027c50375cda040587980fa2e92e985f536/port/dwc2/usb_dc_dwc2.c#L181-L194

- 这里也少了等待条件
[image]

https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/b2b4c45001f1b6b72ae449713607896fd6ca4c6b/Src/stm32h7xx_ll_usb.c#L484-L519

sakumisu, 2024-08-13:
https://github.com/CherryUSB/cherryusb_stm32/blob/master/usb_device/stm32f429igt6/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_ll_usb.c 并没有。请参考 dwc2 官方手册，不要信这些非官方代码

sakumisu, 2024-08-13:
目前这些没有实际影响，所以暂不考虑，只认可 DWC2 官方手册

sakumisu, 2024-08-13:
再给几个参考都是没有的 https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/v1.6.0/Src/stm32h7xx_ll_usb.c

sakumisu, 2024-08-13:
ST 自己都不知道要不要加呢

sakumisu, 2024-08-13:
大佬可以加readme 底下qq群，聊聊

wdfk-prog, 2024-08-13:
> 大佬可以加readme 底下qq群，聊聊

\(^o^)/~好

closed 2024-08-13 by wdfk-prog

wdfk-prog, 2024-08-13:
> 目前这些没有实际影响，所以暂不考虑，只认可 DWC2 官方手册

[image]

- DWC2手册里面是有说要的,觉得最好得补上;
- 能成功的自然不会说有问题,失败的也不一定知道是什么地方出了问题

sakumisu, 2024-08-13:
> ![image](https://private-user-images.githubusercontent.com/66928464/357339138-c7afa520-4ec7-408b-9d2f-722391ba9dba.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

小改动没什么没问题，可以加上

sakumisu, 2024-08-13:
已更新