cherry-embedded/CherryUSB #239: [STM32H7] 使能CONFIG_USB_DWC2_DMA_ENABLE无效,无法传输USB数据

wdfk-prog, 2024-08-15:
- 使用STM32H7,使用USB Device,使用CDC_ACM,虚拟串口

- 不使能`CONFIG_USB_DWC2_DMA_ENABLE`,可以用`wireshark`抓到USB数据
- 使能`CONFIG_USB_DWC2_DMA_ENABLE`,用`wireshark`抓不到任何USB数据
- 所以我认为`CONFIG_USB_DWC2_DMA_ENABLE`在`STM32H7`上无效

- 在[stm32](https://github.com/CherryUSB/cherryusb_stm32)这个仓库中没有找到`CONFIG_USB_DWC2_DMA_ENABLE`的使用

- 是需要外部自行实现DMA相关的配置代码吗?

```log
[I/USB] ========== dwc2 udc params ==========
[I/USB] CID:00002300
[I/USB] GSNPSID:4f54330a
[I/USB] GHWCFG1:00000000
[I/USB] GHWCFG2:229fe190
[I/USB] GHWCFG3:03b8d2e8
[I/USB] GHWCFG4:e3f00030
[I/USB] dwc2 fsphy type:1, hsphy type:2, dma support:2
[I/USB] dwc2 has 9 endpoints and dfifo depth(32-bit words) is 952, default config: 8 endpoints
[I/USB] =================================
[I/USB] fifo0 size:0010, offset:0100
[I/USB] fifo1 size:0080, offset:0110
[I/USB] fifo2 size:0010, offset:0190
[I/USB] fifo3 size:0010, offset:01a0
[I/USB] fifo4 size:0010, offset:01b0
[I/USB] fifo5 size:0010, offset:01c0
[I/USB] fifo6 size:0000, offset:01d0
[I/USB] fifo7 size:0000, offset:01d0
```

sakumisu, 2024-08-15:
请参考文档，谢谢，里面说明的非常清楚，看完再问

sakumisu, 2024-08-15:
对了dwc2移植问题，不要发issue，付费提问

closed 2024-08-15 by sakumisu