cherry-embedded/CherryUSB #244: 外接hub时，可能出现hub定时器停止轮询的情况

qqjmk15263, 2024-08-20:
usbh_hub.c文件中
`static void hub_int_complete_callback(void *arg, int nbytes)
{
    struct usbh_hub *hub = (struct usbh_hub *)arg;

    if (nbytes > 0) {
        usbh_hub_thread_wakeup(hub);
    } else if (nbytes == -USB_ERR_NAK) {
        /* Restart timer to submit urb again */
        USB_LOG_DBG("Restart timer\r\n");
        usb_osal_timer_start(hub->int_timer);
    } else {
        USB_LOG_DBG("Hub other error code:%d\r\n", nbytes);
    }
}`

nbytes 如果出现其他错误值（如 -USB_ERR_IO），轮询定时器将会停止，导致usbhub上的设备状态无法更新，必须重启设备
手动将代码改成
`static void hub_int_complete_callback(void *arg, int nbytes)
{
    struct usbh_hub *hub = (struct usbh_hub *)arg;

    if (nbytes > 0) {
        usbh_hub_thread_wakeup(hub);
    } else if (nbytes == -USB_ERR_NAK) {
        /* Restart timer to submit urb again */
        USB_LOG_DBG("Restart timer\r\n");
        usb_osal_timer_start(hub->int_timer);
    } else {
        USB_LOG_DBG("Hub other error code:%d\r\n", nbytes);
        usb_osal_timer_start(hub->int_timer);
    }
}`
后可正常运行

sakumisu, 2024-08-20:
出错就应该停止使用

qqjmk15263, 2024-08-20:
> 出错就应该停止使用，检查硬件和 USB 时序

我这边测试的时候是偶尔会出现  -USB_ERR_IO  过个1秒钟左右会自己恢复，使用旧版时不会有问题，现在升级到1.3.1，平时正常使用，偶尔出现个错误，USB便无法恢复

closed 2024-08-20 by sakumisu