cherry-embedded/CherryUSB #254: chipidea_reg is actually hpmicro

karlp, 2024-09-20:
The commit https://github.com/cherry-embedded/CherryUSB/commit/2a563dc41fe5e0f57a2f6cca67fbd0a49b51685d introducing chipidea device controller support adds a file: https://github.com/cherry-embedded/CherryUSB/blob/master/port/chipidea/usb_chipidea_reg.h

That file is not actually chipidea register defines, but a hpmicro file.

sakumisu, 2024-09-20:
Hpmicro use chipidea ip.

closed 2024-09-20 by sakumisu

karlp, 2024-09-20:
Ok, to clarify, the file "usb_chipidea_reg.h" is for  the chipidea HS IP, which is EHCI.  The Chipidea FS IP, used on NXP kinetis, MindMotion mm32, Pic32, dsPIC, PIC24, is different, and the one I expected to find under chipidea.  I was expecting that the chipidea EHCI code was in the echi tree.

sakumisu, 2024-10-01:
Support fs ip, device is ok, host is ongoing.