cherry-embedded/CherryUSB #255: Any plans for RP2350 support?

matsobdev, 2024-09-22:
Same as title.

sakumisu, 2024-09-23:
Sorry, no plan for rp series, you can use tinyusb instead.

closed 2024-09-23 by sakumisu

matsobdev, 2025-01-12:
Actually it is working with RP2350. Just required registers configuration change - pulling from SDK for RP2350. Interrupts was mandatory as well.

sidd-kishan, 2025-01-12:
@matsobdev Please share your changes I can port it to the latest version

matsobdev, 2025-01-12:
Tested with SDK 1.5.1 and 2.1.0. I guess it might work with latest release right of the bat, but tried first with safe bet.
[RP2350_changes.zip](https://github.com/user-attachments/files/18390271/RP2350_changes.zip)

sakumisu, 2025-01-17:
> Tested with SDK 1.5.1 and 2.1.0. I guess it might work with latest release right of the bat, but tried first with safe bet. [RP2350_changes.zip](https://github.com/user-attachments/files/18390271/RP2350_changes.zip)
> 
> PS. ARM is working, RISCV is not. If I did `__riscv` definition right, there is still something more to it. PS2. It works as well with 1.4.2 version of CherryUSB (and not with RISCV xD): [RP2350_changes_1.4.2.zip](https://github.com/user-attachments/files/18413917/RP2350_changes_1.4.2.zip). Then `busid` is hardcoded 0, for example:
> 
> ```
> usbd_ep_start_write(0, HID_INT_EP, raport, 6);
> ```
> 
> and `reg_base` - `USBCTRL_REGS_BASE` from Pico SDK:
> 
> ```
> hid_keyboard_init(0, USBCTRL_REGS_BASE);
> ```
> 
> PS3. I'm not sure about `USBCTRL_REGS_BASE` - other garbage works as well.

Thanks, pico official contacts with me and we will restart pico support now.

sakumisu, 2025-01-17:
> [@matsobdev](https://github.com/matsobdev) Please share your changes I can port it to the latest version

Thank you also.

matsobdev, 2025-01-18:
Good news to hear that. I saw some movements in https://github.com/raspberrypi/pico-sdk/issues/1057.